greiman/SdFat #213: No way to disable use of Serial in the library

arkhipenko, 2020-10-11:
Hi,

There is a number of methods in the library that use references to Serial explicitly (e.g., ls() or a number of print methods) which could not be excluded from the compilation. 

While those methods are great and useful during debugging, they prevent the compilation of a production release version with Serial communications completely disabled. 

Is there a way to introduce a compile parameter, similar to `ENABLE_ARDUINO_FEATURES `that excludes any methods with references to Serial from compile scope?

Thank you.
Anatoli

greiman, 2020-10-11:
No, there is only the ENABLE_ARDUINO_FEATURES macro and this was intended for my use of the library with RTOS systems.

arkhipenko, 2020-10-11:
I will buy you a beer if you implement that feature... :)

greiman, 2020-10-12:
I won't implement it.  It's not useful since Serial is not linked unless you call one of the functions in the .h files that use Serial.

I designed it so all calls to Serial are inline.

arkhipenko, 2020-10-12:
OK. I will have to fork it then.
For the STM32 boards there is a way to compile with **no Serial at all**. That is when all references to Serial are showing as errors. 
STM32s are quite popular, so this may come back...
Cheers. 
Anatoli

closed 2020-10-12 by arkhipenko

greiman, 2020-10-12:
SdFat has been forked 322 times and I have had many of request like yours.  [SdFat-beta](https://github.com/greiman/SdFat-beta) has already been forked 52 times and I am just about to make it the release version. 

I use STM32 but with [ChibiOS/RT](https://www.chibios.org/dokuwiki/doku.php?id=start), that's why the ENABLE_ARDUINO_FEATURES is in SdFat.  Arduino is a poor OS for STM32H7.

I started to breakout Arduino features in [SdFat-beta](https://github.com/greiman/SdFat-beta) and it has the following symbols but I only use ENABLE_ARDUINO_FEATURES.  I have not tested the symbols lately.

```
#define ENABLE_ARDUINO_FEATURES 1
#define ENABLE_ARDUINO_SERIAL 1
#define ENABLE_ARDUINO_STRING 1
```

james-portman, 2023-12-19:
This is still an annoying problem

greiman, 2023-12-19:
I no longer am using ChibiOS with SdFat so I only test with Arduino.

Since there is no way to prevent `Serial` from being available in a compile with the Arduino IDE I can't debug this option:

`#define ENABLE_ARDUINO_SERIAL 0`

I know simple programs compile with the above option and Serial is not in the map file for an example like this:

```
#include "SdFat.h"
SdFat sd;
File file;
void setup() {
  sd.begin();
  file.open("test.txt");
}

void loop() {
}
```

I don't know if Serial needs to be defined for the compile.

I know that I get a compile error if I add something like `sd.ls()` that uses Serial.

I am only going to maintain SdFat as a Arduino library.