greiman/SdFat #215: SD library initialises card but not when using SdFat ?

glennpierce, 2020-11-05:
I have some initial test code where I write a simple file using sd.h.

This works fine but I would like to use sdfat directly as I will need to do more advanced things.
The trouble is when I change my code to SdFat the card initialisation fails.

The code is the same in both versions.

```
void initialise_sdcard() {

  Serial.print("Initializing SD card...");

  if (!SD.begin(SD_CS_PIN)) {
    Serial.println("initialization failed!");
    return;
  }
  Serial.println("initialization done.");
}
```

What am I missing ? Does SD.h do something else ?

Thanks

greiman, 2020-11-05:
See [this ](https://github.com/greiman/SdFat/blob/master/examples/ReadWrite/ReadWrite.ino)example.

The big difference is that SdFat uses a high SPI rate.  SD.h uses a low speed.  You can try:
```
#define SPI_SPEED SD_SCK_MHZ(4)
  // ....
  if (!SD.begin(SD_CS_PIN, SPI_SPEED)) {
```

glennpierce, 2020-11-05:
That seems to have done it. Thanks