greiman/SdFat #216: Please explain me two lines

paloscsaba, 2020-11-08:
Hi Bill,

in seekSet, there are two lines I cannot understand:
```

  nCur = (m_curPosition - 1) >> (m_vol->clusterSizeShift() + 9);
  nNew = (pos - 1) >> (m_vol->clusterSizeShift() + 9); 
```

Why do you subtract 1 from m_curPosition and pos?
I suspect, this might induce an error when one of them is at the start of a new block.

Thanks in advance for your explanation,
Csaba

greiman, 2020-11-09:
The current cluster is the cluster that contains the byte before that position.  When the position is the start of a new cluster, the cluster for that position has not been allocated so the value is unknown. For an empty file or a file at position zero, the current cluster is zero, an invalid cluster.

paloscsaba, 2020-11-09:
Oh, this makes sense. Thank you for the explanation.

Grats for the library. I am creating my own one specialized for my own due-based platform (allowing more memory allocation, needing less features) using yours as a basis for reduction, refactoring and change. I am really impressed, till now I have only found one typo that cannot cause problems, but no real bugs. Excellent work, and really thought through to be usable on the smallest platforms!

greiman, 2020-11-09:
I just replaced SdFat V1 with SdFat V2 so there won't be new features in V1.

paloscsaba, 2020-11-10:
Hi Bill,

another question, in FatFile.cpp, write():
```
      // Check for cache sector in write range.
      if (sector <= m_vol->cacheSectorNumber()
          && sector < (m_vol->cacheSectorNumber() + nSector)) {
        // Invalidate cache if cache sector is in the range.
        m_vol->cacheInvalidate();

```
I suspect, the second part of the condition is still incorrect. I saw you fixed already the first part, it was earlier:
```
      if (m_vol->cacheBlockNumber() <= block
          && block < (m_vol->cacheBlockNumber() + nb)) {
        // invalidate cache if block is in cache
        m_vol->cacheInvalidate(); 
```
I would write:
```
      if (sector <= m_vol->cacheSectorNumber()
          && m_vol->cacheSectorNumber() < sector + nSector) {
```
Do you agree?

greiman, 2020-11-10:
Thanks,  it needs to be fixed.