greiman/SdFat #220: Incorrect version/upgrade suggested in platformio

thijstriemstra, 2020-11-14:
Not sure this is the right place to post this but platformio is recommending to upgrade to sdfat v2016.7.24 because it's a higher version nr than 2.0.1 :-/ This means any update in platformio (maybe also arduino ide?) will not be picked up it seems.

[image]

RudyFiero, 2020-11-14:
I am not using PlatformIO. I'm using the standard Arduino IDE. 

SdFat installed is shown as 1.1.2
I select 2.0.1 to be installed.  Select the Install button.  Then it updates the list of libraries. It does not download the new version.

greiman, 2020-11-15:
I had to delete SdFat from Arduino/libraries/ before the library manager would install 2.0.1.   I had a development version 2.0.0 that was never tagged as a release on github.  After I deleted the 2.0.0 version, I could select any version and the library manager would install it.

A similar problem may occur with platformio.  v2016.7.24 was never tagged on GitHub.  I don't use platformio and don't do anything to handle versions on platformio.  I notice platformio lists both version 2.0.0 and 2016.7.24 as releases but I never tagged them as releases on GitHub.

thijstriemstra, 2021-06-03:
Seems fixed now.

closed 2021-06-03 by thijstriemstra