greiman/SdFat #223: Need to recover modification date and time of files and directories

gallegojm, 2020-11-23:
Hi
I am updating my FtpServer library to use your new version of SdFat.
I need to recover the modification date and time of files and can not do it out of SdFat anymore as I was doing with older version.
I solved it adding those three routines (based on existing routines **printModifyDateTime**() )
In **FatFile** class:
```
bool FatFile::getModifyDateTime( uint16_t * pdate, uint16_t * ptime ) {
  DirFat_t dir;
  if (!dirEntry(&dir)) {
    DBG_FAIL_MACRO;
    goto fail;
  }
    * pdate = getLe16( dir.modifyDate );
    * ptime = getLe16( dir.modifyTime );
  return true;
fail:
  return false;
}

```
In **ExtFatFile** class:
```
bool ExFatFile::getModifyDateTime( uint16_t * pdate, uint16_t * ptime ) {
  DirFile_t* df = reinterpret_cast<DirFile_t*>
                 (m_vol->dirCache(&m_dirPos, FsCache::CACHE_FOR_READ));
  if (!df) {
    DBG_FAIL_MACRO;
    goto fail;
  }
    * pdate = getLe16( df->modifyDate );
    * ptime = getLe16( df->modifyTime );
  return true;
fail:
  return false;
}

```
In **FsBaseFile** class (inline inFsFile.h)
```
  bool getModifyDateTime( uint16_t * pdate, uint16_t * ptime ) {
    return m_fFile ? m_fFile->getModifyDateTime(pdate,ptime) :
           m_xFile ? m_xFile->getModifyDateTime(pdate,ptime) : false;
  }

```
Is it possible you add those routines in your next release?
Or to give me an idea of how to solve it out of the library?

Thank you for your library

greiman, 2020-11-23:
I will provide these functions in the next release:
```
bool getAccessDateTime( uint16_t * pdate, uint16_t * ptime );
bool getCreateDateTime( uint16_t * pdate, uint16_t * ptime );
bool getModifyDateTime( uint16_t * pdate, uint16_t * ptime );
```
FAT32/FAT16 only has access date so I will return zero.

gallegojm, 2020-11-23:
Thank you so much!