greiman/SdFat #224: Difference between versions 1.1.4 and 2.0.1

gallegojm, 2020-11-26:
Hi
Unless I am mistaken, when creating a file, without using the date/time callback function:
- version 1.1.4 set modification date and time to 2000-01-02 00:00 (as printed by printModifyDateTime)
but
- version 2.0.1 (with SDFAT_FILE_TYPE set to 1) set modifyDate and modifyTime to 0 in the directory entry (printModifyDateTime print nothing)
Is it intentional?
This can be solved using the callback function, but I think it would be better to maintain compatibility, unless there is an other reason.

greiman, 2020-11-26:
Setting date/time fields to zero was intentional.  The FAT standard says if date and time fields are not supported, they should be set to 0 on file create and ignored on other file operations.  Later versions say modify date and time must be supported but other fields may be zero.

Looks like I need to put a valid date time in the modify fields.  Windows dir and Linux ls do strange things with zero.

I looked at [FatFs](http://elm-chan.org/fsw/ff/00index_e.html) and Chan uses 00:00:00 January 1 of the current year for the modify date time.

I looked at 1.1.4 and and default date/time is 2000-01-01 01:00.
```
/** Default date for file timestamps is 1 Jan 2000 */
const uint16_t FAT_DEFAULT_DATE = ((2000 - 1980) << 9) | (1 << 5) | 1;
/** Default time for file timestamp is 1 am */
const uint16_t FAT_DEFAULT_TIME = (1 << 11);
```
I think I will follow Chan, and use YYYY-01-01 00:00 where YYYY is the release year or maybe the compile year.  I will leave the create and access fields zero.

gallegojm, 2020-11-26:
> Windows dir and Linux ls do strange things with zero.

Yes, and some programs too.
For example FtpRush disconnect from server when listing a directory with files with zero in modify date time.
Any date, 2000 or like Chan's solution, would fix this.
Thank you for your help.

greiman, 2020-12-01:
I added new code for default time stamps to the latest SdFat-beta version 2.0.2-beta.3.  The values are defined in SdFatConfig.h.  I will soon move this beta to the release version.
```
/**
 * Set the default file time stamp when a RTC callback is not used.
 * A valid date and time is required by the FAT/exFAT standard.
 *
 * The default below is YYYY-01-01 00:00:00 midnight where YYYY is
 * the compile year from the __DATE__ macro.  This is easy to recognize
 * as a placeholder for a correct date/time.
 *
 * The full compile date is:
 * FS_DATE(compileYear(), compileMonth(), compileDay())
 *
 * The full compile time is:
 * FS_TIME(compileHour(), compileMinute(), compileSecond())
 */
#define FS_DEFAULT_DATE FS_DATE(compileYear(), 1, 1)
/** 00:00:00 midnight */
#define FS_DEFAULT_TIME FS_TIME(0, 0, 0)
```

gallegojm, 2020-12-02:
Hi
Thank you for the fast update for setting and recovering date and time to files.
My first test of FtpServer looks to work fine with SdFat-beta version 2.0.2-beta.3
At least with sd card using Fat16/32 files system.
I need to purchase a card of larger capacity to try exFat, althougt not having the utility of files so large as 4 Giga

greiman, 2020-12-02:
Thanks for the feedback.  It helps me decide when to move a beta to the release repo.