greiman/SdFat #226: V2 fails with boot sector in sector 0

MitchBradley, 2020-12-03:
I have an old 4GB SD card that has a boot sector, not an MBR, in sector 0.  It works with V1.1.3 but fails with V2.0.1.  I think I see where the problem lies:

v1.1.3 FatVolume.h has
```
  bool init() {
    return init(1) || init(0);
  }
```
That tries the boot sector case  - init(0 -  if the MBR attempt - init(1) - fails.

It works with this change to v2.0.1 FatVolume.h:
```
   bool begin(BlockDevice* dev, bool setCwv = true, uint8_t part = 1) {
-    if (!init(dev, part)) {
+    if (!init(dev, part) && !init(dev, 0)) {
```

(Environment: platformio, ESP32 Arduino framework)

greiman, 2020-12-03:
I removed support for no MBR since the SD standard has not supported it since File System Specification Version 3.00 in 2009.

Modern SD cards depend on a strict layout of the files system with structures in specified places.  This allows performance improvements since the card caches things like FAT entries in a sector cache and data regions in a larger cache.  SD cards have extremely large flash pages, often 32KiB, so 512 bytes sectors are emulated by RAM caches in the SD.

You can use the [Standard SD formatter](https://www.sdcard.org/downloads/formatter/) or the the SdFat SdFormatter example to correctly format the card.

From the SD formatter page:
> The SD Memory Card Formatter formats SD Memory Card, SDHC Memory Card and SDXC Memory Card (respectively SD/SDHC/SDXC Cards) complying with the SD File System Specification created by the SD Association (SDA).
> 
> It is strongly recommended to use the SD Memory Card Formatter to format SD/SDHC/SDXC Cards rather than using formatting tools provided with individual operating systems. In general, formatting tools provided with operating systems can format various storage media including SD/SDHC/SDXC Cards, but it may not be optimized for SD/SDHC/SDXC Cards and it may result in lower performance.

MitchBradley, 2020-12-03:
I hope you might be persuaded to reconsider. I understand the desire to stay up to date, but backwards compatibility is a huge advantage from a customer support standpoint.  As a developer and support person for Grbl_Esp32, we field a lot of support requests for people having trouble with SD cards not working.  Having to tell them things like "try another card" or "reformat your brand new card with a different partition map and put FAT32 in it, instead of the factory ExFAT" is a big time sink for both parties.  We were hoping that switching to SdFat would decrease our support problems.  They might, but if we are trading one set of problems for another, the choice becomes less clear.

Was the support for boot sectors on legacy cards causing other problems?  Obviously the base code is still there and can be enabled with a lightweight change.

I certainly would not recommend formatting a modern card with a legacy layout - and indeed, formatters often go to some lengths to disallow that.  But telling customers that they must reformat an existing, otherwise fully-functional, card - and save and restore the data that was on it - seems like a poor customer experience.  Especially in light of the fact that it is not at all easy to discover exactly what the problem was.  It is like the old joke where the first thing the support person suggests is to reinstall the operating system.  Easier said than done.

greiman, 2020-12-03:
I have had wasted lots of time with cards formatted in nonstandard ways, mostly by Linux users.  A common problem is to format with very small clusters and complain about performance.  There are more fundamental problems with reliability with power loss during wear leveling remaps when the file structure are not placed as intended.

I now have the policy that I don't support nonstandard formats.

No legacy cards were manufactured without a MBR.  Many early cards were reformatted without a MBR using utilities written by third parties.  Card like your 4GB SD had 8192 sectors before the FAT boot sector so you could gain 4MB by reformatting. 

Finally the SD Association provided a formatter.

I checked my  February 2000 copy of "File System Specification Version 1.0"  and it specifies:

"For the identification of the Data Area as a partition, the first sector has Master Boot Record and Partition Table.  And the SD Memory Card file system uses the FAT file system (ISO/IEC 9293) and supports both FAT12 and FAT16 as the file system type."

I dug out a 17 year old 16 MB FAT12 card that has write protect set and has never been written. It has a MBR.  I bought and saved these cards so I could reverse engineer the file layout.  Now I have full access to full SD Association specs.

MitchBradley, 2020-12-03:
I cannot prove it, but I really think that the card in question was factory-formatted.  It is a SanDisk Extreme III SDHC card.  The reason I think it was factory formatted is because I have a deep aversion to reformatting these things with any tool whatsoever, for the reasons you have already mentioned, particularly concerns that the manufacturer's Flash Translation Layer firmware depends on quirks of the layout.  Reformatting with Linux tools is particularly dangerous.  I discovered that the hard way when I was the firmware lead for the One Laptop Per Child project.  Their SD cards (we also supported the MMC variant) carried an ext2 filesystem which could sometimes interact badly with magical factory features such as hidden wear leveling.

Despite one's best efforts, it is hard to avoid dealing with dodgy formats, particularly due to the popularity of Raspberry Pi disk images.

closed 2020-12-03 by MitchBradley

MitchBradley, 2020-12-03:
As an aid to customer support, would you consider adding some way to extract details about why a card failed to mount?  Enabling DBG_FAIL_MACRO works for developers but is not so good for end-user use.  I imagine a feature whereby failure code numbers could be added to a list - possibly in the form of an array. When a top-level operation like .begin() fails, one could invoke, say, .errorList() to extract that list.  The reason for a list instead of a single code is to capture the call-chain context - this low level fail caused the next level up to fail, etc.  I think that could be done without causing a code size explosion.

greiman, 2020-12-05:
I plan to work on the SD init and file system mount code.  I will provide more error information.  If I require an MBR, I can improve the process since the MBR has the partition type.  Now I try exFAT then FAT32/FAT16 and if one succeeds I return true. 

You can get a good idea of the error cause by looking at the hardware error code.  If there is not a hardware error, its a format problem.

Here are my error print routines for an init error:

```
  /** Print error details after begin() fails.
   *
   * \param[in] pr Print destination.
   */
  void initErrorPrint(Print* pr) {
    pr->println(F("begin() failed"));
    if (sdErrorCode()) {
      pr->println(F("Do not reformat the SD."));
      if (sdErrorCode() == SD_CARD_ERROR_CMD0) {  // Card does not respond to reset command.
        pr->println(F("No card, wrong chip select pin, or wiring error?"));
      }
    }
    errorPrint(pr);
  }
  //----------------------------------------------------------------------------
  /** %Print SD errorCode and errorData.
   *
   * \param[in] pr Print destination.
   */
  void errorPrint(print_t* pr) {
    if (sdErrorCode()) {
     // SD hardware problem.
      pr->print(F("SdError: 0X"));
      pr->print(sdErrorCode(), HEX);
      pr->print(F(",0X"));
      pr->println(sdErrorData(), HEX);
    } else if (!Vol::fatType()) {
      pr->println(F("Check SD format."));  // Important info if no hardware problem.
    }
  }
```