greiman/SdFat #227: What is the new, correct way to open file by index in a vwd()?

Jerware, 2020-12-06:
I'm updating some old code that used an ancient version of SdFat, and sometimes I would load files & directories by index in the volume working directory like this:

```
SdFat sd;
SdFile myFile;
fileIndex = 0;
sd.chdir("/00system");
myFile.open(sd.vwd(), fileIndex, O_READ);

```
I'd check to see if the file isDir(), and go from there. With the new version of SdFat, I get the following error:

**error: 'SdFat {aka class SdFs}' has no member named 'vwd'**

I tried searching for a solution but have had no luck. I would very much appreciate knowing how to update my code. Thank you.

greiman, 2020-12-06:
SdFs doesn't have a vwd file.  FsVolume is a wrapper for either a exFat volume or a FAT volume.

I am not sure what functionality you need.   Can you open the path in chdir something like this?
```
SdFat sd;
SdFile myFile;
SdFile dir;
fileIndex = 0;
dir.open("/00system", O_READ);
sd.chdir("/00system");
myFile.open(&dir, fileIndex, O_READ);
```

Jerware, 2020-12-06:
I see! Thank you.

closed 2020-12-06 by Jerware