greiman/SdFat #230: Async reading/writing

fabiuz7, 2020-12-20:
Hi,
 I would try to avoid the typical sporadic slow down of SD cards (10ms or even more), so I have tried to find alternatives to the "default" Arduino SD library. I have tried to dig your library, but I'm not sure if it supports asynchronous (non-blocking) operations. I read `BufferedPrint` example, but it seems to improve overall performance, but not avoid sporadic slow operations.
Does your library provide a way to avoid such latency?

greiman, 2020-12-21:
It is possible to avoid the long delays on write by buffering and checking for SD busy.  See at about line 368 of the [ExFatLogger](https://github.com/greiman/SdFat/blob/master/examples/ExFatLogger/ExFatLogger.ino).  This program can log data at 4000 samples per second on a fast ARM board.

To read without long delays works best with dedicated SPI and a contiguous exFAT card. The other solution is to use a RTOS like [ChibiOS ](https://github.com/greiman/ChRt)and a read ahead thread.

greiman, 2020-12-21:
I need a simple solution for writes with low latency.  I will modify the BufferedPrint class to have an option to use the buffering to minimize latency by timing the writes to when the card  is not busy.

greiman, 2020-12-22:
You can probably make the BufferedPrint class perform well for latency by adding a sync() call to your program after writing.

You should use a large buffer,  up to 256 bytes. If you are logging data periodically, add this code after writing data.  This will write to the SD when latency will be low.

```
if (!sd.card()->isBusy()) {
  bp.sync();
}
```
I need to make the buffer index uint16_t so larger buffers can be used and make the index public so you only call flush when more than some amount of data is in the buffer.