greiman/SdFat #233: Can't initialize SD card on STM32 (worked on 1.1.4, need translation?)

MockbaTheBorg, 2020-12-22:
Hi Greiman,

First of all, happy holidays.
I have this app I wrote based on SdFat called RunCPM, which is a Z80 CP/M emulator.
When using STM32 boards I used to initialize the SD like this:
```
SdFatSoftSpiEX<PC8, PD2, PC12> SD; // MISO, MOSI, SCK
#define SDINIT PC11 // CS

  if (SD.begin(SDINIT)) {
    // do something.
  }
```

However once I migrated to 2.0.2 SdFatSoftSpiEX is not there, and I have not yet found a way to translate that.
If I do:
```
SdFat SD;
#define SDINIT PC11 // CS
```
It builds fine but won't detect the presence of the card. I am missing the part where I specify the pins I guess, but I still can't figure out how it would be done now.
All my other platforms are working fine, teensy, ESP32, Arduino Due. Only the STM32 eludes me.
Based on the aforementioned method of using the SD card, what would be the right way to use it now?

Thanks a lot in advance.

greiman, 2020-12-22:
There is not an easy way to use the STM32 SDIO driver.  It would need a wrapper to present the API that the Teensy SDIO driver presents.

SdFatSoftSpiEX is just the DEDICATED_SPI option in V2.  So use the same options as the [SoftwareSpi](https://github.com/greiman/SdFat/blob/master/examples/SoftwareSpi/SoftwareSpi.ino) example.

I assume there is no SPI driver available for the pin connected to the SD.  The SPI pins are CMD, D0, D3 and clk. 

What board are you using?  Are you using the Arduino STM32 core package by STMicroelectronics?

MockbaTheBorg, 2020-12-22:
I am using this STM32F4DISCOVERY: https://www.st.com/resource/en/user_manual/dm00039084-discovery-kit-with-stm32f407vg-mcu-stmicroelectronics.pdf

With this LCD adapter for the SD card: http://www.kentecdisplay.com/uploads/soft/Products_spec/EB-STM32F4DISCOVERY-LCD_UserGuide_V01.pdf

I opted for the SPI because I wasn't able to find a way to use SDIO back in the day. SPI should be fine I guess.

MockbaTheBorg, 2020-12-22:
Good news ... by using the method from the SoftwareSpi I was able to access the card. All I needed to do was to replace the pin names.
Thank you!!! You can close this one.
Thanks a lot and happy holidays.