greiman/SdFat #235: Go back to numeric versions

Steve132, 2020-12-24:
https://github.com/greiman/SdFat/blob/1535ac2b0332c22da26ca876fd2a04641dffadb4/src/SdFat.h#L42

Previous versions of SdFat prior to v2.X had versioning done with an integer, which makes preprocessor checks possible. e.g. `#define SD_FAT_VERSION 10102`

This allows code such as 

    #ifdef SD_FAT_VERSION > 10100 
    ...
    #else
    #error
    #endif
    
However, the new "string" version makes this impossible, because "" evaluates to a const pointer in C, which then [cannot be compared or evaluated in the preprocessor](https://stackoverflow.com/questions/2335888/how-to-compare-strings-in-c-conditional-preprocessor-directives).  At all.

The fix would be to maintain compatibility with the old way and have an additional preprocessor for the string for printing.

E.g.

     #define SD_FAT_VERSION 20200
     #define SD_FAT_VERSION_STR "2.0.2"
     
It's important to do this asap before people begin adoption of v2

greiman, 2020-12-25:
I will go to an integer version.  It won't make the preprocessor compare entirely reliable since there was no consistent format before.

I started using a data before the Arduino library manager. 
```
/** SdFat version YYYYMMDD */
#define SD_FAT_VERSION 20150718
```
Then I used a string for a while. 
```
/** SdFat version */
#define SD_FAT_VERSION "1.0.14"
```

The last integer version for V1 was not proper.  For 1.1.4 I didn't update the version.
```
/** SdFat version 1.1.2 */
#define SD_FAT_VERSION 10102
```
I will attempt to use a 5 digit version in the future.  where 2.1.3 becomes 20103.

greiman, 2020-12-26:
See release 2.0.3.  I used your suggestion.
```
/** SdFat version  for cpp use. */
#define SD_FAT_VERSION 20003
/** SdFat version as string. */
#define SD_FAT_VERSION_STR "2.0.3"
```