greiman/SdFat #236: please add "const" qualifier to ExFatFile::fileSize() getter

elhayra, 2020-12-26:
Hi,

Since this function is a simple getter, it should be const for safety and so it gets compiled with client const functions

greiman, 2020-12-26:
I have quickly gone through .h files and added const to over 50 simple things like fileSize(). I will look for more cases where I need to change both the .h and .cpp files and then publish a new SdFat-beta to see what it breaks.  So in a week or two you will see a const fileSize().

I stopped paying attention to const after years of never hearing from users.  Good thing for me to start worrying about const again.

greiman, 2020-12-30:
Try 2.0.4 for  improved "const correctness".

elhayra, 2021-01-01:
thank you for the quick response.
closing.

closed 2021-01-01 by elhayra