greiman/SdFat #238: stm32F4 SPI1 support?

vitasam, 2020-12-29:
Hello, 
I'm trying `SdInfo` example with stm32F411CEU6 ("black-pill") board, but can't get SD card working with SPI1. I've been trying different options from `SdFatConfig.h` and hardcoded sketch to:
`#define SD_CONFIG SdSpiConfig(PA4, SHARED_SPI, SD_SCK_MHZ(16), &SPI)`  but there is no effect.

The standard Arduino SD library works smothly with HW SPI1.

Is there any chance to get `SdFat` working with HW SPI?

Thank you! Happy New Year!

greiman, 2020-12-30:
What STM32 package are you using? This [one](https://github.com/rogerclarkmelbourne/Arduino_STM32) or this [one](https://github.com/stm32duino/Arduino_Core_STM32)  or another one?

vitasam, 2020-12-30:
Just checked from Arduino's _Additional Board Manager URLs_: the 2nd one: `https://github.com/stm32duino/BoardManagerFiles/raw/master/STM32/package_stm_index.json`

greiman, 2020-12-30:
How do you setup for SPI1?  Can you post the code to setup and call sd.begin?

vitasam, 2020-12-30:
I did not change much in [SdInfo](https://github.com/greiman/SdFat/tree/master/examples/SdInfo) example (mostly commented out the stuff) and added a clock setting function (looks like the clock settings are wrong in Arduino stm32 core):

```
#define SD_CONFIG SdSpiConfig(PA4, SHARED_SPI, SD_SCK_MHZ(16), &SPI)
...
```

```
void SystemClock_Config(void)
{
  RCC_OscInitTypeDef RCC_OscInitStruct = {0};
  RCC_ClkInitTypeDef RCC_ClkInitStruct = {0};

  /** Configure the main internal regulator output voltage
  */
  __HAL_RCC_PWR_CLK_ENABLE();
  __HAL_PWR_VOLTAGESCALING_CONFIG(PWR_REGULATOR_VOLTAGE_SCALE1);
  /** Initializes the RCC Oscillators according to the specified parameters
  * in the RCC_OscInitTypeDef structure.
  */
  RCC_OscInitStruct.OscillatorType = RCC_OSCILLATORTYPE_HSE;
  RCC_OscInitStruct.HSEState = RCC_HSE_ON;
  RCC_OscInitStruct.PLL.PLLState = RCC_PLL_ON;
  RCC_OscInitStruct.PLL.PLLSource = RCC_PLLSOURCE_HSE;
  RCC_OscInitStruct.PLL.PLLM = 25;
  RCC_OscInitStruct.PLL.PLLN = 192;
  RCC_OscInitStruct.PLL.PLLP = RCC_PLLP_DIV2;
  RCC_OscInitStruct.PLL.PLLQ = 4;
  if (HAL_RCC_OscConfig(&RCC_OscInitStruct) != HAL_OK)
  {
    //Error_Handler();
    return;
  }
  /** Initializes the CPU, AHB and APB buses clocks
  */
  RCC_ClkInitStruct.ClockType = RCC_CLOCKTYPE_HCLK|RCC_CLOCKTYPE_SYSCLK
                              |RCC_CLOCKTYPE_PCLK1|RCC_CLOCKTYPE_PCLK2;
  RCC_ClkInitStruct.SYSCLKSource = RCC_SYSCLKSOURCE_PLLCLK;
  RCC_ClkInitStruct.AHBCLKDivider = RCC_SYSCLK_DIV1;
  RCC_ClkInitStruct.APB1CLKDivider = RCC_HCLK_DIV2;
  RCC_ClkInitStruct.APB2CLKDivider = RCC_HCLK_DIV1;

  if (HAL_RCC_ClockConfig(&RCC_ClkInitStruct, FLASH_LATENCY_3) != HAL_OK)
  {
    //Error_Handler();
    return;
  }
}
```

greiman, 2020-12-30:
I don't see where you setup SPI1.  You use SPI.

> I'm trying SdInfo example with stm32F411CEU6 ("black-pill") board, but can't get SD card working with SPI1.

`#define SD_CONFIG SdSpiConfig(PA4, SHARED_SPI, SD_SCK_MHZ(16), &SPI)`

In most Aduino board packages SPI is SPI0, the first port and SPI1 is the second SPI port.

greiman, 2020-12-30:
Does Sdinfo work unmodified on the standard SPI port?  If I use a Nucleo 446RE with the SD connected to the same pins as a 328 Arduino, SdInfo works unmodified.  Here is the output.

```
type any character to start
init time: 2 ms

Card type: SDHC

Manufacturer ID: 0X73
OEM ID: BG
Product: NCard
Version: 1.0
Serial number: 0XC2111004
Manufacturing date: 5/2012

cardSize: 4015.00 MB (MB = 1,000,000 bytes)
flashEraseSize: 128 blocks
eraseSingleBlock: true

OCR: 0XC0FF8000

SD Partition Table
part,boot,bgnCHS[3],type,endCHS[3],start,length
1,0X0,0X2,0X3,0X1,0XB,0X38,0XF8,0XCC,8192,7833600
2,0X0,0X0,0X0,0X0,0X0,0X0,0X0,0X0,0,0
3,0X0,0X0,0X0,0X0,0X0,0X0,0X0,0X0,0,0
4,0X0,0X0,0X0,0X0,0X0,0X0,0X0,0X0,0,0

Scanning FAT, please wait.

Volume is FAT32
sectorsPerCluster: 64
clusterCount:      122272
freeClusterCount:  62
fatStartSector:    14472
dataStartSector:   16384
```

vitasam, 2020-12-30:
Good point. I will double check it.

vitasam, 2020-12-30:
SdInfo does not work unmodified on SPI1 port (PA4, PA5,PA6,PA7) on F411CEU6 "black-pill" board from Ali

greiman, 2020-12-30:
Looks like you are on your own, I tried it on a STM32F411 Nucleo and it all works with no mod to anything.  Just put a SD shield on the Nucleo and select Nucleo F411RE as the board part number.

vitasam, 2020-12-30:
Could you kindly double check what are SD card pins on your Nucleo F411 board?

greiman, 2020-12-30:
D13 PA5 SPI1_SCK
D12 PA6 SPI1_MISO
D11 PA7 TIM1_CH1N or SPI1_MOSI
D10 PB6 TIM4_CH1 or SPI1_CS

See page 44 of [this](https://www.st.com/resource/en/user_manual/dm00105823-stm32-nucleo64-boards-mb1136-stmicroelectronics.pdf).

vitasam, 2020-12-30:
Hmm, interesting. If I'm lucky, what I need to do is to connect SD card CS to PB6 and SdInfo example should work out of the box:
[image]

vitasam, 2020-12-30:
Nope, SdInfo did not work unmodified:
```
21:03:55.633 -> SdFat version: 2.0.4
21:03:55.633 -> 
21:03:55.633 -> Assuming the SD is the only SPI device.
21:03:55.633 -> Edit DISABLE_CS_PIN to disable an SPI device.
21:03:55.633 -> 
21:03:55.633 -> Assuming the SD chip select pin is: 41
21:03:55.633 -> Edit SD_CS_PIN to change the SD chip select pin.
21:03:55.633 -> 
21:03:55.633 -> type any character to start
21:03:58.775 -> 
21:03:58.775 -> SD initialization failed.
21:03:58.775 -> Do not reformat the card!
21:03:58.775 -> Is the card correctly inserted?
21:03:58.775 -> Is there a wiring/soldering problem?
21:03:58.775 -> Is SD_CS_PIN set to the correct value?
21:03:58.775 -> Does another SPI device need to be disabled?
21:03:58.775 -> SD errorCode: SD_CARD_ERROR_CMD0 = 0x1
21:03:58.775 -> SD errorData = 0xff
21:03:58.775 -> 
21:03:58.775 -> type any character to start
```

vitasam, 2020-12-30:
Wow! It works! I just defined the SS pin:
`#define SDCARD_SS_PIN  PB6`

```
21:13:36.470 -> SdFat version: 2.0.4
21:13:36.470 -> 
21:13:36.470 -> Assuming the SD is the only SPI device.
21:13:36.470 -> Edit DISABLE_CS_PIN to disable an SPI device.
21:13:36.470 -> 
21:13:36.470 -> Assuming the SD chip select pin is: 22
21:13:36.470 -> Edit SD_CS_PIN to change the SD chip select pin.
21:13:36.470 -> 
21:13:36.470 -> type any character to start
21:13:41.257 -> init time: 222 ms
21:13:41.257 -> 
21:13:41.257 -> Card type: SD2
21:13:41.257 -> 
21:13:41.257 -> Manufacturer ID: 0X3
21:13:41.257 -> OEM ID: SD
21:13:41.257 -> Product: SU01G
21:13:41.257 -> Version: 8.0
21:13:41.257 -> Serial number: 0X4C04DC01
21:13:41.257 -> Manufacturing date: 10/2009
21:13:41.257 -> 
21:13:41.257 -> cardSize: 988.28 MB (MB = 1,000,000 bytes)
21:13:41.257 -> flashEraseSize: 32 blocks
21:13:41.257 -> eraseSingleBlock: true
21:13:41.257 -> 
21:13:41.257 -> OCR: 0X80FF8000
21:13:41.257 -> 
21:13:41.257 -> SD Partition Table
21:13:41.257 -> part,boot,bgnCHS[3],type,endCHS[3],start,length
21:13:41.257 -> 1,0X0,0X0,0X28,0X0,0XE,0XFE,0X3F,0X77,39,1929248
21:13:41.257 -> 2,0X0,0X0,0X0,0X0,0X0,0X0,0X0,0X0,0,0
21:13:41.257 -> 3,0X0,0X0,0X0,0X0,0X0,0X0,0X0,0X0,0,0
21:13:41.257 -> 4,0X0,0X0,0X0,0X0,0X0,0X0,0X0,0X0,0,0
21:13:41.257 -> 
21:13:41.257 -> Scanning FAT, please wait.
21:13:41.592 -> 
21:13:41.592 -> Volume is FAT16
21:13:41.592 -> sectorsPerCluster: 32
21:13:41.592 -> clusterCount:      60273
21:13:41.592 -> freeClusterCount:  60269
21:13:41.592 -> fatStartSector:    47
21:13:41.592 -> dataStartSector:   551
21:13:41.592 -> Data area is not aligned on flash erase boundary!
21:13:41.592 -> Download and use formatter from www.sdcard.org!
21:13:41.592 -> 
21:13:41.592 -> type any character to start
```

vitasam, 2020-12-31:
Thank you! Happy New Year!

closed 2020-12-31 by vitasam