greiman/SdFat #241: The SD initialization failed

loliconce, 2021-01-03:
I do not know how to deal with this problem.
The error code:

SD errorCode: SD_CARD_ERROR_CMD0 = 0x1
SD errorData = 0xff

greiman, 2021-01-03:
Try running the [QuickStart](https://github.com/greiman/SdFat/blob/master/examples/QuickStart/QuickStart.ino) example.  Reply with the results as printed, a description of your board, SD module, wiring, and any other useful info.

Post the results like this:
```
SPI pins:
MISO: 12
MOSI: 11
SCK:  13
SS:   10

Be sure to edit DISABLE_CHIP_SELECT if you have
a second SPI device.  For example, with the Ethernet
shield, DISABLE_CHIP_SELECT should be set to 10
to disable the Ethernet controller.

SD chip select is the key hardware option.
Common values are:
Arduino Ethernet shield, pin 4
Sparkfun SD shield, pin 8
Adafruit SD shields and modules, pin 10

Enter the chip select pin number: 5

Assuming the SD is the only SPI device.
Edit DISABLE_CHIP_SELECT to disable another device.

SD initialization failed.
Do not reformat the card!
Is the card correctly inserted?
Is chipSelect set to the correct value?
Does another SPI device need to be disabled?
Is there a wiring/soldering problem?

errorCode: 0x1, errorData: 0xff

Restarting

Enter the chip select pin number: 
```

loliconce, 2021-01-04:
@greiman 

The info：

SPI pins:
MISO: 12
MOSI: 11
SCK:  13
SS:   10

Be sure to edit DISABLE_CHIP_SELECT if you have
a second SPI device.  For example, with the Ethernet
shield, DISABLE_CHIP_SELECT should be set to 10
to disable the Ethernet controller.

SD chip select is the key hardware option.
Common values are:
Arduino Ethernet shield, pin 4
Sparkfun SD shield, pin 8
Adafruit SD shields and modules, pin 10

Enter the chip select pin number: 4

Assuming the SD is the only SPI device.
Edit DISABLE_CHIP_SELECT to disable another device.

SD initialization failed.
Do not reformat the card!
Is the card correctly inserted?
Is chipSelect set to the correct value?
Does another SPI device need to be disabled?
Is there a wiring/soldering problem?

errorCode: 0x1, errorData: 0x9

Restarting

Enter the chip select pin number:

greiman, 2021-01-04:
Can't help you. You didn't give me info I asked for.  

> Reply with the results as printed, a description of your board, SD module, wiring, and any other useful info.

Do you have the SD module connected to the required pins?

```
SPI pins:
MISO: 12
MOSI: 11
SCK: 13
Enter the chip select pin number: 4
```

loliconce, 2021-01-04:
@greiman 

board  :Arduino UNO 
SD module: https://chinese.alibaba.com/product-detail/SD-Card-Module-Slot-Socket-Reader-62484062968.html

I did connect the SD module to the require pins.

greiman, 2021-01-04:
The board will never work.  It does not have level shifters to convert from the Uno 5 V signals to the SD 3.3 V requirement.

loliconce, 2021-01-04:
@greiman 

What about this one :
https://shopee.ph/Micro-SD-Card-Module-i.18252381.255216867?gclid=EAIaIQobChMIlrfLqKGC7gIVI4ZLBR0mkADgEAQYAiABEgIlgPD_BwE

greiman, 2021-01-04:
You need a module like[ this](https://www.ebay.com/itm/1Pc-Micro-SD-Storage-Board-Mciro-SD-TF-Card-Memory-Shield-Module-SPI-For-Arduino/352697431787).  The part labled U1 is the level shifter.

loliconce, 2021-01-04:
@greiman
👍 Thank you !

greiman, 2021-01-04:
Look like it should work.  It has a level shifter.