greiman/SdFat #243: FatFile->getName return bool but FsBaseFile->getName expects size_t

saschagu, 2021-01-07:
The implementation of FsBaseFile->getName returns a size_t (length of the filename):
https://github.com/greiman/SdFat/blob/cb5a7d2afb4eb94f2563044645fc816f0ea4c2c8/src/FsLib/FsFile.h#L213
and returns the getName function of the FatFile member variable:
https://github.com/greiman/SdFat/blob/cb5a7d2afb4eb94f2563044645fc816f0ea4c2c8/src/FsLib/FsFile.h#L215
Opposed to the ExFatFile version of getName FatFile->getName returns a bool:
https://github.com/greiman/SdFat/blob/cb5a7d2afb4eb94f2563044645fc816f0ea4c2c8/src/FatLib/FatFile.h#L336

greiman, 2021-01-08:
This is one of many SdFs return value bugs.  I am working on these, missing functions, and "const correctness" issues.  Soon a beta will be released with these fixes and some new features.  It be maybe several weeks before this beta makes it to release.

saschagu, 2021-01-08:
OK, thanks for your fast answer and your great work! Waiting for the next release!