greiman/SdFat #245: What would be the right way to init on the Teensy 4.1, compared to Teensy 3.6

MockbaTheBorg, 2021-01-13:
Hi there,

I have an app which runs on the Teensy 3.6 and is refusing to detect the card on the Teensy 4.1.
On the teensy v3.6 I initialize like this:
```
SdFat32 SD;
typedef File32 File;
#define SDINIT SdioConfig(FIFO_SDIO)
  if (SD.begin(SDINIT)) {
...
```
On the Teensy v4.1 I have tried the same as above, as well as:
```
SdFat SD;
#define SDINIT SdioConfig(FIFO_SDIO)
  if (SD.begin(SDINIT)) {
...
```
But none seems to work.
I actually had it working some time ago (can't remember when), but since then I have upgraded Arduino IDE as well as SdFat from 2.0.2 to 2.0.4.
No when recompiling for all the supported platforms, Teensy v4.1 is refusing to initialize the card.
Wondering what would be the correct way to initialize it so I could start eliminating variables.

Also, is the definition of SDMHZ at all relevant to FIFO_SDIO?

Thanks a lot,
Marcelo.

greiman, 2021-01-13:
I do lots of development with Teensy 4.1 and have no problems, I almost always use exactly what you are using.  I assume the SD card you are using is not formatted exFAT.

Use of some libraries on Teensy causes the typedef for File to not be defined.  Apparently File is defined in the include file FS.h.  Also on ARM boards SdFat is defined as SdFs which works for both exFAT and FAT16/FAT32.

Add a call to sd.initErrorHalt() like this:
```
// Use Teensy FIFO SDIO
#define SD_CONFIG  SdioConfig(FIFO_SDIO)
  ...
  if (!sd.begin(SD_CONFIG)) {
    sd.initErrorHalt(&Serial);
  }
```
Post the printout for the failure.

MockbaTheBorg, 2021-01-14:
Hi Greiman,

The initErrorHalt printed:
```
begin() failed
Do not reformat the SD.
SdError: 0X17,0X0
```
I believe it is something related to this card specifically. I have multiple cards of the same brand, some "KingMAX" 256MB.
They all work on all devices I have, Teensy 3.5, Teensy 3.6, Teensy 4.0 (with the audio board SD), some ESP32 boards, etc.
They refuse to work only on the Teensy 4.1 specifically.
I have just tried a 4GB card on the 4.1 and it worked. So we can consider it to be a card problem, but I would still be curious to find out the reason, so I can avoid certain types of cards in the future? I dunno.

Thanks,
Marcelo.

greiman, 2021-01-14:
The error is a timeout in SdFat/src/SdCard/SdioTeensy.cpp at about line 620.  It may be that Teensy 4.1 runs the polling loop faster than Teensy 3.6.  The card is trying to initialize, setting up data structures and buffers.  Your card is probably a really old design since it is a 256MB card and may be overwhelmed by the intense polling. 

Here are two things to try in this loop.
```
  do {
    if (!cardAcmd(0, ACMD41_XFERTYP, arg) ||
       ((micros() - m) > BUSY_TIMEOUT_MICROS)) {
      return sdError(SD_CARD_ERROR_ACMD41);
    }
  } while ((SDHC_CMDRSP0 & 0x80000000) == 0);
```
Try replacing BUSY_TIMEOUT_MICROS with 4000000UL, currently it is 1000000.
Try adding a delay at the bottom of the loop, maybe 100 ms.

Let me know what happens.  

If these ideas fail, there is little that can be done since the command is done by microcode int SDHC controller.  

The Teensy 4 controller is a new design that can support really fast cards.  Paul has not enabled the 1.8V signaling required for 100MB/sec cards but the controller is capable of those rates.

MockbaTheBorg, 2021-01-14:
Cool. I will give it a try just so we know of that would be the issue.
If that doesn't work then we chalk this one as "unsupported card", as it is
indeed old ones.

MockbaTheBorg, 2021-01-14:
Yep ... no combination of the above helped. So we should assume that this card won't work with the v4.1 (at least not by using SDIO, as it works on the v4.0 via SPI).
Thanks a lot for the help, you can mark this one as closed.