greiman/SdFat #246: SoftwareSpi  not working in 2.0.4

patricklaf, 2021-01-18:
Hello,
I am trying to port a sketch which uses SdFat, from version 1.1.4 to latest version 2.0.4.
I am using an Arduino Mega 2560 and a MCUFRIEND_kbv TFT shield.

This is the working code for 1.1.4:
```cpp
SdFatSoftSpi<12, 11, 13> fat;
...
fat.begin(10);
```
Then the code for 2.0.4
```cpp
// In SdFatConfig.h
//
// #define SPI_DRIVER_SELECT 2

const uint8_t SD_CS_PIN = 10;
const uint8_t SOFT_MISO_PIN = 12;
const uint8_t SOFT_MOSI_PIN = 11;
const uint8_t SOFT_SCK_PIN  = 13;

SoftSpiDriver<SOFT_MISO_PIN, SOFT_MOSI_PIN, SOFT_SCK_PIN> softSpi;

#define SD_CONFIG SdSpiConfig(SD_CS_PIN, DEDICATED_SPI, SD_SCK_MHZ(0), &softSpi)

SdFat fat;
...
if (!fat.begin(SD_CONFIG)) {
        Serial.print(F("error code: 0x"));
        Serial.println(fat.card()->errorCode());
        Serial.print(F("error data: 0x"));
        Serial.println(fat.card()->errorData());
        fat.initErrorHalt();
}

```
This is the output of the code:
```
error code: 0x0
error data: 0x0
begin() failed
Check SD format.
```
I am using a 2 Gb FAT32 SDCARD.

greiman, 2021-01-18:
I can't help since I don't have your hardware.  

When I use a Mega 2560 and an old Adafruit Data logging shield which connects to SPI pins CS = 10, MISO = 12, MOSI = 11, SCK = 13, the SoftwareSpi example runs without modification. 
```
Type any character to start
This line was printed using software SPI.
Done.
```
I used a 2GB FAT32 card.

Try reformatting the card with the [SD formatte](https://www.sdcard.org/downloads/formatter/)r from the SD Association or the [SdFat example](https://github.com/greiman/SdFat/tree/master/examples/SdFormatter).  The new SdFat requires a format with a MBR as an aid to determine the card type.

patricklaf, 2021-01-18:
You helped me. More than that, you solved the problem.
I formatted my SD with SD formatter (curiously, it is now FAT16). 
It's OK now, the sketch is running as before with the 1.4.1 library.
Thanks a lot.

closed 2021-01-18 by patricklaf

greiman, 2021-01-18:
I should have caught the FAT32 format.  I used a FAT16 card since the standard is FAT16 for cards 128MB through 2 GB.