greiman/SdFat #247: "BackwardCompatability.ino" works with SD.h, not with SdFat on STM32F103

FrankM54, 2021-01-20:
(First of all, THANK YOU for all of your work on SdFat, I appreciate it)

On the latest version of SdFat (v2.0.4) there is an example program that allow you to switch between SD.h and SdFat.h for SD card support using SPI. 

I can't get SdFat to work, but SD.h works! 
I had the same behaviour with version v1.1.4 and updated to v2.0.4 without resolving the issue.

Setup Details:
• STM32F103C8 BluePill clone
• Official ST Core
• Arduino IDE 1.8.13
• MacOS 10.14.6
• No bootloader installed
• Use Serial connection and STM32CubeProgrammer to program
• SD module wired to SPI1 (SS = PA4, SCK = PA5, MISO = PA6, and MOSI = PA7)
• SdFat v2.0.4

Setting USE_SD_H to 1 uses SD.h. In this example, the SD card works without any issues:
`Using SD.h. Set USE_SD_H zero to use SdFat.h.`
`Type any character to begin.`
`Initializing SD card...initialization done.`
`Writing to test.txt...done.`
`test.txt:`
`testing 1, 2, 3.`
`testing 1, 2, 3.`
`testing 1, 2, 3.`
`testing 1, 2, 3.`

However, when I set USE_SD_H to 0 to use SdFat and it doesn't work:
`Using SdFat.h. Set USE_SD_H nonzero to use SD.h.`
`Type any character to begin.`
`Initializing SD card...initialization failed! `

I am using a full size SD card reader, wired to SPI1, without a level shifter, BUT since the STM32 is 3.3v a level shifter is not needed. Also, since the SD.h programs work, the SD card and wiring is confirmed to be correct.

After much testing, all the SD.h examples work, but I can't get any SdFat.h example to work.
Is there anything I can do to resolve this?

greiman, 2021-01-20:
It may be the SPI speed.  I have seen problems before with Blue Pill clones.

SD.h uses a 4 MHz rate.

Try changing the begin line like this:
`   if (!SD.begin(SD_CS_PIN, SD_SCK_MHZ(4) )) {`

FrankM54, 2021-01-20:
Thank you for the quick reply!

I will try your suggestion shortly, but I found some thing else interesting...
I used a 8 channel logic analyzer to capture the SPI bus using SD.h and SdFat.h

After power on, the SPI clock is 280Khz and enters the native operating mode / get ready / waits for commands.
While using SD.h, after SPI mode is selected CS goes high, then low and the CLK is now 2.2 Mhz. The text is written to the card and read back. Total bus activity time is 134.4 ms.

After searching for a bit, I  saw the text "testing 1, 2, 3." (which was interesting to see it from the wire!!)

With SdFat.h I think an error is being reported from the card, (still digging in to SD command & error documentation) as the CLK stays at 280 Khz. Total bus activity time is 6.54 ms.

I will try your suggestion (and capture it to see what is different) next.

Here are some screen shots of the start up.
SD.h start up:
[image]

SdFat.h start up:
[image]

FrankM54, 2021-01-20:
Wow, you are REALLY good!

Changing `if (!SD.begin(SD_CS_PIN)) {`
to `if (!SD.begin(SD_CS_PIN, SD_SCK_MHZ(4) )) {`
resolves the issue! - Thank you!

I did capture the SPI bus and will have a look at it for my own education and I am curious to see what the logic analyzer sees as the CLK speed. Also, I will update my post on arduino.cc with the solution.

FrankM54, 2021-01-20:
Quick note: 
After setting the SPI clock speed to `SD_SCK_MHZ(4)` the logic analyzer still shows it at 2.286 Mhz (I suspect it really is 72/32 = 2.25 Mhz), which is the same as before. Weird!

Anyways, thanks for the quick resolution.

greiman, 2021-01-20:
The specification for SPISettings(speedMaximum, dataOrder, dataMode) is that you request speedMaximum and the SPI library produces the highest speed the board can deliver that is less than or equal to the requested speed.

By default I ask for the highest speed the card can support which is always 50 MHz for modern cards.  I would expect STM32F103 chips to support 36 MHz on SPI1 but for some reasons the quality on many BluePill boards causes failures.

The ST libraries work fine with other STM32F103 boards and run at 36 MHz.  I suspect you could set speed to 18 MHz.

The slow initialization speed is part of the SD standard and dates to 1997 when the first MMC cards appeared.  This insures you can identify a card then reject it since the slow speed MMC cards were FAT12 and I only support FAT12 if you ask for it in SdFatConfig.h

FrankM54, 2021-01-20:
Understood and Agreed. 
I spent the day reading the SD Assoc "Physical Layer Simplified  Spec v8", and then decoding what was on the SPI bus. I was curious to understand what was going on and the low level commands.

As you know, it starts off CMD0, CMD8, CMD55, CMD41 and finishes the slow frequency section with a CMD58.
Then the clock frequency switches to a higher freq (in my case to ~2.2xx Mhz) followed by the set up and data transfer.
(I still need to decode that section)

Here is the CLK frequency transition point with:
 'if (!SD.begin(SD_CS_PIN, SD_SCK_MHZ(4) )) {'
[image]

Here is what happens with out the fix or
'if (!SD.begin(SD_CS_PIN, SD_SCK_MHZ(50) )) {'
[image]

It seems like the STM32F1 couldn't switch to the higher SPI CLK frequency.
Again, thanks for your quick help with this :)

FrankM54, 2021-01-21:
Minor Update:
With my setup, increasing SD_SCK_MHZ() up to 18Mhz also worked very well and increased speed.

closed 2021-01-21 by FrankM54