greiman/SdFat #249: Migrating from V1 to V2 missing createContigious

Bambofy, 2021-01-25:
Hi, i'm am migrating my project from V1 to V2 SDFat, previously i was using SdBaseFile and SdFatEX to set up a contiguous file:

```
SdBaseFile myFile;
SdFatEX mySd;
bool contigFileOK = myFile.createContiguous("my file name", 1);
```

However the createContiguous function is not declared in the new SdFs and FsFile classes, how can i replicate these functions with the new version?

Cheers!
Richard

greiman, 2021-01-25:
Use open() followed by preAllocate().  I prefer this sequence since you then know what failed, bad open parameters or no contiguous free space and you can reopen and truncate a file.

```
  // open or create file - truncate existing file.
  if (!file.open("bench.dat", O_RDWR | O_CREAT | O_TRUNC)) {
    error("open failed");
  }
  if (!file.preAllocate(FILE_SIZE)) {
      error("preAllocate failed");
  }
```

Bambofy, 2021-01-25:
> Use open() followed by preAllocate(). I prefer this sequence since you then know what failed, bad open parameters or no contiguous free space and you can reopen and truncate a file.
>
> ```
>   // open [...]

Cheers! i'll try this out