greiman/SdFat #252: a bug found for the stm32 (libmaple) stm32-duino cores (stm32f4, stm32f1)

ag88, 2021-02-06:
https://github.com/greiman/SdFat/blob/master/src/SpiDriver/SdSpiSTM32.cpp#L59
```
uint8_t SdSpiArduinoDriver::receive(uint8_t* buf, size_t count) {
#if USE_STM32_DMA
  //return m_spi->dmaTransfer(nullptr, buf, count);
  return m_spi->dmaTransfer((const uint16_t) 0xff, buf, count);

#else  // USE_STM32_DMA
  m_spi->read(buf, count);
  return 0;
#endif  // USE_STM32_DMA
}
```
the call to dmaTransfer should use the dmaTransfer((const uint16_t) 0xff, buf, count) call.
using the nullptr call in the F4 (libmaple) (roger's, steve's) stm32duino core 
causes a bug and no data is read from the card

the issue and case is discussed here, the sketch tested is the SdInfo example
https://www.stm32duino.com/viewtopic.php?f=7&t=911
SdInfo failed on all cards i happen to have when i tested it without the patch and all works once i did this patch
note that i've also discussed the api differences in spi.h, spi.cpp api between f1 and f4 (libmaple cores).
i think it is likely we'd use that same as f1 returning an int8_t error code (0 - success) 
hopefully that'd reduce some if-defs

ag88, 2021-02-06:
there are some other off-topic observations
https://github.com/greiman/SdFat/blob/master/src/SdCard/SdSpiCard.cpp#L371
```
// send command and return error code.  Return zero for OK
uint8_t SdSpiCard::cardCommand(uint8_t cmd, uint32_t arg) {

  // send CRC - correct for CMD0 with arg zero or CMD8 with arg 0X1AA
  spiSend(cmd == CMD0 ? 0X95 : 0X87);
#endif  // USE_SD_CRC
```
that 0x95, 0x87 is for the crc for CMD0 and CMD8. a rather strange observation is that cardCommand() is also called by readRegister. that would make it look strange as the crc for other card commands won't give the same crc as 0x95, 0x87.
but interestingly it didn't cause a problem with 'incorrect' crc for the several cards i have. Hence, ignored for now. i'm not sure if using 0 in place of that funky crc would actually cause any 'new' problems.

ag88, 2021-02-06:
another off topic initially i encountered this strange dependency error
../../SdFat/FsLib/FsFile.h:397:13: error: 'FsVolume' has not been declared
i got around it putting
``` #include "../../SdFat/FsLib/FsVolume.h" ``` in FsFile.h
i'm not too sure if it may be my build stack that is broken

greiman, 2021-02-06:
I forgot to remove libmaple driver.  I am only supporting the official ST board package.

closed 2021-02-06 by greiman