greiman/SdFat #253: rmRfStar() is not available for FsFile

roener, 2021-02-10:
Thanks for this cool library!

I'm migrating from v1 to v2 and working on a SAMD device.
Is there a technical limitation why `rmRfStar()` is not available for `FsFile`? It looks like, it is only implemented in `FatFile` class.

greiman, 2021-02-10:
I would be possible to implement rmRfStar() for exFAT.  I originally implemented rmRfStar() since there was no function to format an SD.

I found that many users really needed a format function so format is available for FAT16/FAT32, exFAT and SdFs.  Format insures the volume is not corrupt and the root directory is clean.

I probably should have an example that shows how to delete selected files in a directory with openNext().

roener, 2021-02-11:
Thanks for your quick reply!

Actually I need to delete a directory with many logging files in it. Other files in the root directory must be kept.
So rmRfStar() came in handy for this task.
Thank you for pointing me to openNext(). With this, I can easily implement what I need. 👍

closed 2021-02-11 by roener