greiman/SdFat #259: Compile errors in FatPartition and FsCache

slabbi, 2021-02-20:
libraries\SdFat\common\FsCache.cpp.o (symbol from plugin): In function `FsCache::sync()':
(.text+0x0): multiple definition of `FsCache::sync()'
libraries\SdFat\ExFatLib\ExFatPartition.cpp.o (symbol from plugin):(.text+0x0): first defined here
libraries\SdFat\common\FsCache.cpp.o (symbol from plugin): In function `FsCache::sync()':
(.text+0x0): multiple definition of `FsCache::get(unsigned long, unsigned char)'
libraries\SdFat\ExFatLib\ExFatPartition.cpp.o (symbol from plugin):(.text+0x0): first defined here
d:\eigene dateien\arduino\libraries\sdfat\src\common\fscache.h:38:7: note: type name 'FsCache' should match type name 'FatCache'
 class FsCache {
       ^
D:\Eigene Dateien\Arduino\libraries\SdFat\src\FatLib\FatPartition.h:68:7: note: the incompatible type is defined here
   /** Create an instance of FatPartition
       ^
D:\Eigene Dateien\Arduino\libraries\SdFat\src/FatLib/FatVolume.h:38:7: note: type 'struct FatVolume' itself violates the C++ One Definition Rule
 class FatVolume : public  FatPartition {
       ^
D:\Eigene Dateien\Arduino\libraries\SdFat\src\FatLib\FatVolume.h:38:7: note: the incompatible type is defined here
 class FatVolume : public  FatPartition {
       ^
collect2.exe: error: ld returned 1 exit status
exit status 1
Fehler beim Kompilieren für das Board ATmega2560.

slabbi, 2021-02-20:
Was fine in 2.0.2, errors appear in 2.0.3 and 2.0.4

closed 2021-02-20 by slabbi

greiman, 2021-02-20:
The Arduino IDE is really fragile.  It tries too hard to optimize builds.  When I test new versions I often need to clear temporary files when I try various options.  I have wasted many hours so I understand.