greiman/SdFat #260: volumeBegin failed. Is the card formatted?

simeon9696, 2021-02-20:
I'm using the SdInfo.ino sketch and I've formatted the 1GB micro sd card with the [official formatter](https://www.sdcard.org/downloads/formatter/). Here's the full output of the sketch: 
```
SdFat version: 2.0.4

Assuming the SD is the only SPI device.
Edit DISABLE_CS_PIN to disable an SPI device.

Assuming the SD chip select pin is: 15
Edit SD_CS_PIN to change the SD chip select pin.

type any character to start
init time: 17 ms

Card type: SD2

Manufacturer ID: 0X0
OEM ID: 0␀
Product: APPSD
Version: 0.0
Serial number: 0X40030000
Manufacturing date: 12/2011

cardSize: 998.24 MB (MB = 1,000,000 bytes)
flashEraseSize: 128 blocks
eraseSingleBlock: true

OCR: 0X80FF8000

SD Partition Table
part,boot,bgnCHS[3],type,endCHS[3],start,length
1,0XAD,0X14,0XAF,0X5D,0XEA,0X97,0XA9,0X6F,681453721,1853796126
2,0X5A,0XED,0X62,0X84,0X5C,0X78,0XB9,0X49,612340677,1951551534
3,0X22,0XBE,0XD0,0X76,0XE5,0XEA,0XA9,0X18,1796606188,778673004
4,0XFD,0X72,0XA5,0XB1,0X48,0X51,0X87,0X80,1494544675,2599889341

MBR not valid, assuming Super Floppy format.

volumeBegin failed. Is the card formatted?

type any character to start
```

I am using an ESP8266 with the following wire config:
```
 * D5   <-> SCK/CLK
 * D6   <-> MISO
 * D7   <-> MOSI
 * D8   <-> CS
 * Vcc  <-> Vcc
 * Gnd  <-> Gnd
```

And I changed line 25 from `const uint8_t SD_CS_PIN = SS;` to `const uint8_t SD_CS_PIN = D8;`

How can I fix the volumeBegin failed error? If I use the `SdFormatter.ino`, the sketch works but then Windows says 'you need to format the disk to use it' when I reinsert it into the machine.

greiman, 2021-02-21:
I can't help.  The partition table reads as junk.  

I would need to look at your setup with a logic analyzer but I don't have you hardware and card.

simeon9696, 2021-02-21:
Thanks @greiman I tried a 64GB card from SanDisk and I got a print out similar to yours above. So I guess the wiring etc is good, just the cards are bad?