greiman/SdFat #262: Can't find a valid FAT16/FAT32 partition

Twilight-Logic, 2021-02-24:
I am having problems with this SdFat library, the primary problem being it not recognizing the FAT32 partition on the SD Card. I am doing a preliminary test using the QuickStart example.

The first problem I encountered is that I couldn't enter '10' as the chip select pin number. The sketch accepts only 1 (i.e. a single character) and then fails. I ended up commenting out the prompt (lines 108-117) and simply substituting a manual sssignment:

chipSelect = 10;

That allowed the program to continue to the next stage, however, the program then failed to read the FAT32 volume details. This is the output:

```
PI pins:
MISO: 50
MOSI: 51
SCK:  52
SS:   53

Be sure to edit DISABLE_CHIP_SELECT if you have
a second SPI device.  For example, with the Ethernet
shield, DISABLE_CHIP_SELECT should be set to 10
to disable the Ethernet controller.

SD chip select is the key hardware option.
Common values are:
Arduino Ethernet shield, pin 4
Sparkfun SD shield, pin 8
Adafruit SD shields and modules, pin 10

Assuming the SD is the only SPI device.
Edit DISABLE_CHIP_SELECT to disable another device.

Card successfully initialized.
Can't find a valid FAT16/FAT32 partition.
Try reformatting the card.  For best results use
the SdFormatter program in SdFat/examples or download
and use SDFormatter from www.sdcard.org/downloads.

Restarting
```

Again, if I comment out the two following returns to allow the program execution to carry on,  I get:

```
Card successfully initialized.

Card size: 32212 MB (MB = 1,000,000 bytes)

Volume is FAT0, Cluster size (bytes): 0

Files found (date time size name):

This card should be reformatted for best performance.
Use a cluster size of 32 KB for cards larger than 1 GB.
Only cards larger than 2 GB should be formatted FAT32.
Try reformatting the card.  For best results use
the SdFormatter program in SdFat/examples or download
and use SDFormatter from www.sdcard.org/downloads.

Restarting
```

The volume is identified and directoris/files read just fine using the standard CardInfo example from the SD library. The card also reads fine on a Windows 10 machine.

My setup is an Arduino Mega 2560 with a standalone MicroSD Card adapter that one can readily obtain on eBay. The SD Card is an Adata 32Gb SDHC card. For good measure, I re-formatted the card using the FAT type on Windows 10. Unlike Windows 7 which has options for FAT16 and FAT32, Windows 10 only seems to have one option: FAT and formats as FAT32.

SdFat library version is 2.0.4.

Thanks.

Twilight-Logic, 2021-02-24:
Just a further update. I followed the advice to reformat the card with the SDFormatter program from sdcard.org and the blank volume is now being identified. The question now is, why does formatting the card on Windows 10 cause volume detection to fail with SdFat, but works fine with the standard Arduino SD library?

greiman, 2021-02-24:
The standard SD library uses the first version of SdFat that I wrote in 2008.  At that time many SD cards were being reformatted with a non-standard format that did not have a MBR partition table and I allowed it.  Modern cards are designed to only support the standard SD format.

I now only allow the standard format.  Problems were occurring with many cards, especially with Linux formatted cards. 

The SD association specifies the layout of the files system including hidden sectors, cluster size etc.  The structure of a card's flash, buffers, and algorithms depends on not just one partition but the location of file system structures and cluster sizes.

You should never use an OS utility to reformat an SD card.  Use the [SD Association formatter](https://www.sdcard.org/downloads/formatter/) .

Here is a quote from the SD Association.

> This specification specifies SD file systems. The type of the file system to be used shall be uniquely decided with Card Capacity as follows. Here, Card Capacity means the total size of Data Area and Protected Area size.  
> 
> File System Type That is, all Standard Capacity SD Memory Cards whose capacity is 2048MB or less shall use FAT12 / FAT16 file system, and never use the other file system. Similarly, all High Capacity SD Memory Cards shall use FAT32 file system, and never use the other file system. And all Extended Capacity SD Memory Cards shall use exFAT file system, and never use the other file system. This includes the prohibition of partial format of SD Memory Card. For example, 8GB High Capacity SD Memory Card should not be formatted as 2GB card with FAT12 / FAT16 file system. In this case, whole area of 8GB should be formatted with FAT32 file system.

I don't know why you can't enter 10 in the QuickStart example.  Are you using the Arduino Serial Monitor?

Twilight-Logic, 2021-02-24:
Thank you for the information regarding the SD Association formatter. I will stick with that.
Whereas it would not surprise me for Microsoft to do their own thing, your report of problems with cards formatted on Linux did did surprise me.

Regarding the ability to enter 10, no, I was using PuTTy. It works fine in the Serial Monitor.
Thanks.

greiman, 2021-02-24:
PuTTy is great and it sends a character at a time which is proper.

Unfortunately Serial Monitor is popular, it is "line at a time" and many people use it with "No line ending" so I just wait a fraction of a second for Serial data.  I don't like the Arduino Stream solution of waiting one second and I hate to force NL, CR or NL & CR.

Linux allows a huge number of options for vfat so novice users try almost anything.

> mkfs.vfat [ -a ] [ -A ] [ -b sector-of-backup ] [ -c ] [ -l filename ] [ -C ] [ -f number-of-FATs ] [ -F FAT-size ] [ -h number-of-hidden-sectors ] [ -i volume-id ] [ -I ] [ -m message-file ] [ -n volume-name ] [ -r root-dir-entries ] [ -R number-of-reserved-sectors ] [ -s sectors-per-cluster ] [ -S logical-sector-size ] [ -v ] device [ block-count ]