greiman/SdFat #263: How to format exFAT with different cluster sizes?

Bob128x, 2021-02-24:
Hello,

I would like to know how to setup/format exFAT with different cluster size/allocation size?

Also, looks like formating SD card on windows with cluster size/allocation size of 2048 bytes does not let create files on the SD card.
As far as I can tell it only happens at 2048 bytes. Could you please check this?

Regards

greiman, 2021-02-25:
The SD standard specifies the cluster size for all SD cards. 

The SD association specifies the layout of the files system including hidden sectors, cluster size etc.  The structure of a card's flash, buffers, and algorithms depends on not just one partition but the location of file system structures and cluster sizes.

You should never use an OS utility to reformat an SD card.  Use the [SD Association formatter](https://www.sdcard.org/downloads/formatter/) .

Here is a quote from the SD Association.

> This specification specifies SD file systems. The type of the file system to be used shall be uniquely decided with Card Capacity as follows. Here, Card Capacity means the total size of Data Area and Protected Area size.  
> 
> File System Type That is, all Standard Capacity SD Memory Cards whose capacity is 2048MB or less shall use FAT12 / FAT16 file system, and never use the other file system. Similarly, all High Capacity SD Memory Cards shall use FAT32 file system, and never use the other file system. And all Extended Capacity SD Memory Cards shall use exFAT file system, and never use the other file system. This includes the prohibition of partial format of SD Memory Card. For example, 8GB High Capacity SD Memory Card should not be formatted as 2GB card with FAT12 / FAT16 file system. In this case, whole area of 8GB should be formatted with FAT32 file system.

Bob128x, 2021-02-25:
Thanks for answer Greiman. 

I agree with you, but how I can use your SdFat Lib to try different cluster sizes. (just only for test purpose)

Okey, if I understand correct you use ExFatLib/ExFatFormatter.cpp

```
  // Format exFAT if larger than 32GB.
  bool rtn = cardSectorCount > 67108864 ?
    exFatFormatter.format(m_card, sectorBuffer, &Serial) :
    fatFormatter.format(m_card, sectorBuffer, &Serial);
```

Here are some difinition: 

```
// Formatter assumes 512 byte sectors.
const uint32_t BOOT_BACKUP_OFFSET = 12;
const uint16_t BYTES_PER_SECTOR = 512;
const uint16_t SECTOR_MASK = BYTES_PER_SECTOR - 1;
const uint8_t  BYTES_PER_SECTOR_SHIFT = 9;
const uint16_t MINIMUM_UPCASE_SKIP = 512;
const uint32_t BITMAP_CLUSTER = 2;
const uint32_t UPCASE_CLUSTER = 3;
const uint32_t ROOT_CLUSTER = 4;
```

Just need to know, are these fixed values and/or are they linked somehow to something? 
Or am I on the wrong track?

greiman, 2021-02-25:
I don't have time to waste so I am closing this.  You will need to find the exFAT spec and write your own code.

closed 2021-02-25 by greiman