greiman/SdFat #266: Program memory usage seems high

Twilight-Logic, 2021-02-27:
I initially tried this library due to the amount of dynamic memory used by the standard Arduino SD library. It was useful that the SdFat library used some 10-20% less. Unfortunately, this appears to be more than compensated for by the rather extravagant use of program memory.

The QuickStart example for instance, produces the following results on a Uno:

```
Sketch uses 22938 bytes (71%) of program storage space. Maximum is 32256 bytes.
Global variables use 1060 bytes (51%) of dynamic memory, leaving 988 bytes for local variables. Maximum is 2048 bytes.
```
The sketch is less than 200 lines of code. The "SdInfo" sketch is a little less hungry:

```
Sketch uses 13996 bytes (43%) of program storage space. Maximum is 32256 bytes.
Global variables use 901 bytes (43%) of dynamic memory, leaving 1147 bytes for local variables. Maximum is 2048 bytes.
```

Both of these small example programs use considerably more program memory than the comparable "CardInfo" example from the SD library which produces the following statistic:

```
Sketch uses 10952 bytes (33%) of program storage space. Maximum is 32256 bytes.
Global variables use 1422 bytes (69%) of dynamic memory, leaving 626 bytes for local variables. Maximum is 2048 bytes.
```

I could maybe live with 43% (or 10% overhead) but 71% for a small example program? What is it within the SdFat library that has such a large program memory overhead? With only a handful of routines I was up to 120% of memory usage in my sketch in no time, while the same sketch using the SD library used 84%.

greiman, 2021-02-27:
SD.h uses a 2009 version of SdFat that only supported FAT16/FAT32 with short file names.  By default SdFat V2 supports long filenames.

The QuickStart and SdInfo are configured to use FAT16/FAT32 and exFAT.

If you configure SdFat like SD.h and only use FAT16/Fat32, you can reduce memory use.

Edit SdFatConfig.h and change these two symbols.
```
#if defined(__AVR__) && FLASHEND < 0X8000
// 32K AVR boards.
#define ENABLE_DEDICATED_SPI 0  //1   <<-- change from 1 to 0
#else  // defined(__AVR__) && FLASHEND < 0X8000
```
`#define USE_LONG_FILE_NAMES 0  //1  <<--- change from 1 to 0`

test with this SD.h example:
```
/*
  SD card read/write

 This example shows how to read and write data to and from an SD card file
 The circuit:
 * SD card attached to SPI bus as follows:
 ** MOSI - pin 11
 ** MISO - pin 12
 ** CLK - pin 13
 ** CS - pin 4 (for MKRZero SD: SDCARD_SS_PIN)

 created   Nov 2010
 by David A. Mellis
 modified 9 Apr 2012
 by Tom Igoe

 This example code is in the public domain.

 */

#include <SPI.h>
#include <SD.h>

File myFile;

void setup() {
  // Open serial communications and wait for port to open:
  Serial.begin(9600);
  while (!Serial) {
    ; // wait for serial port to connect. Needed for native USB port only
  }

  Serial.print("Initializing SD card...");

  if (!SD.begin(4)) {
    Serial.println("initialization failed!");
    while (1);
  }
  Serial.println("initialization done.");

  // open the file. note that only one file can be open at a time,
  // so you have to close this one before opening another.
  myFile = SD.open("test.txt", FILE_WRITE);

  // if the file opened okay, write to it:
  if (myFile) {
    Serial.print("Writing to test.txt...");
    myFile.println("testing 1, 2, 3.");
    // close the file:
    myFile.close();
    Serial.println("done.");
  } else {
    // if the file didn't open, print an error:
    Serial.println("error opening test.txt");
  }

  // re-open the file for reading:
  myFile = SD.open("test.txt");
  if (myFile) {
    Serial.println("test.txt:");

    // read from the file until there's nothing else in it:
    while (myFile.available()) {
      Serial.write(myFile.read());
    }
    // close the file:
    myFile.close();
  } else {
    // if the file didn't open, print an error:
    Serial.println("error opening test.txt");
  }
}

void loop() {
  // nothing happens after setup
}
```
With SD.h 
```
Sketch uses 10652 bytes (33%) of program storage space. Maximum is 32256 bytes.
Global variables use 996 bytes (48%) of dynamic memory, leaving 1052 bytes for local variables. Maximum is 2048 bytes.
```
Edit the sketch to use SdFat V2
```
//#include <SD.h> <<-- remove
#include "SdFat.h"  // <<-- add
SdFat SD;              // <<-- add
```
result is:
```
Sketch uses 9026 bytes (27%) of program storage space. Maximum is 32256 bytes.
Global variables use 991 bytes (48%) of dynamic memory, leaving 1057 bytes for local variables. Maximum is 2048 bytes.
```

Twilight-Logic, 2021-02-28:
Thank you for your detailed reply.

Initially I temporarily removed three template functions which reduced the requirement from 120% to 102%. I then tried your suggestions which reduced the requirement further down to 96%. At this point the sketch could be compiled.

I then realised I was still following the example and using 'SdFs sd' and 'ExFile file' definitions so I changed these to 'SdFat32 sd' and 'File32 file' which reduced the program memory requirement further to 69%. This makes the situation more manageable although obviously with restrictions on file naming.

I now have to figure out what to do about the template functions. The main reason for using them was to have a way to print to something other than the default hardware serial port, for example to a CDC serial or SoftwareSerial port. I perhaps need to have a look at your streaming technique using ArduinoOutStream.

Thanks.

greiman, 2021-02-28:
In the above ReadWrite example the new SdFat with short file names take 9,026 bytes of flash while SD.h takes 10,652. SdFat V2 is 1,626 or more than 10% smaller than SD.h.  This is the normal situation for most apps.