greiman/SdFat #268: undefined reference to `vtable for SdioCard'

stevstrong, 2021-02-27:
After I set the option `HAS_SDIO_CLASS` to 1, during compiling the Quickstart.ino example for use with SPI I get the following linker warning message:

```
C:\Users\Steve\AppData\Local\Temp\arduino_build_421836\sketch\QuickStart.ino.cpp.o: In function `_GLOBAL__sub_I_sd':

C:\Users\Steve\Documents\Arduino\libraries\SdFat\examples\QuickStart/QuickStart.ino:187: warning: undefined reference to `vtable for SdioCard'
```

If I comment out the option, then the warning message is gone.

However, in my applications I use alternatively SPI or SDIO, or both. It is not feasible to enable/disable that option every time I do not use SDIO.

Is there a way how to get rid of the warning without editing each time the SDIO option?

greiman, 2021-02-27:
I have no idea why you get the warning.

You can't set HAS_SDIO_CLASS non-zero unless the driver exists because of this class:

```
class SdCardFactory {
 public:
  /** Initialize SPI card.
   *
   * \param[in] config SPI configuration.
   * \return generic card pointer.
   */
  SdCard* newCard(SdSpiConfig config) {
    m_spiCard.begin(config);
    return &m_spiCard;
  }
  /** Initialize SDIO card.
   *
   * \param[in] config SDIO configuration.
   * \return generic card pointer or nullptr if SDIO is not supported.
   */
  SdCard* newCard(SdioConfig config) {
#if HAS_SDIO_CLASS
    m_sdioCard.begin(config);
    return &m_sdioCard;
#else  // HAS_SDIO_CLASS
    (void)config;
    return nullptr;
#endif  // HAS_SDIO_CLASS
  }

 private:
#if HAS_SDIO_CLASS
  SdioCard m_sdioCard;
#endif  // HAS_SDIO_CLASS
  SdSpiCard m_spiCard;
};
```

stevstrong, 2021-02-27:
Thank your for the hint, indeed, the driver was missing, I compiled for STM32F103 chip which did not have any enabled SDIO driver in my core.
I changed the config header to enable SDIO only for F4 family, and the warning was gone.

stevstrong, 2021-02-28:
I come back to this because I have observed an undesired effect: both SDIO and SPI libraries are linked to the project even if I use only one of them. Can you confirm?

Beside making the Sd declaration universal, is there any other technical reason why you merged both SPI and SDIO instances into one single object? The SdioFat instance of V1 was working independent from SPI.

stevstrong, 2021-02-28:
Sorry to bother regarding this, but the warning message is generated again even for F4 family if I set `HAS_SDIO_CLASS` non-zero and use the SPI configuration in begin() instead of SDIO. The software is working correctly in this case, just that the warning message is generated.
Any idea how that can be removed?

greiman, 2021-02-28:
I have no idea what is causing your problems.  I designed V2 so you could edit SdFatConfig.h and have SdFat and File support any combination of file types, only FAT, only, exFAT, or both FAT and exFAT.  and SdFat would support SPI, SDIO, or USB MSC.

I now have four people/groups doing their own major ports/changes and want help.  I am telling all do what ever you want to modify SdFat but don't ask me for help.

I have many people who think I would be excited about their app and want help using V2 with exFAT.  People want me to port SdFat V2 to non-arduino compatible boards.

My email is overflowing so you are on you own.

greiman, 2021-03-01:
Steve I assumed you know what a vtable is.  It's the dispatch table for a class with virtual members.  Any non-pure virtual function must be defined, even if it's never used. A missing definition will often result in the 'vtable undefined' linker error.