greiman/SdFat #269: #error SPI_DRIVER_SELECT must be two in SdFat/SdFatConfig.h

BigBear178, 2021-02-28:
Hello,

I don't know how to get the "SoftwareSpi" example to configure. I downloaded it directly off your site and I added it to my library but I can't get over this issue.  I thought that I messed up the code when I tried to designate my pins but this issue exists for me right when I open the example. I am not great with coding so I apologize if this is something so simple to resolve.

greiman, 2021-02-28:
Edit SdFat/src/SdFatConfig.h and set SPI_DRIVER_SELECT to 2.

BigBear178, 2021-02-28:
Hello, 

I set the SPI_DRIVER_SELECT to 2 but it still doesn't compile correctly. I included a PNG where I changed it. Is this the correct location or...? I appreciate the help!

[image]

greiman, 2021-03-01:
I don't have a clue what your doing so I can't help.

closed 2021-03-01 by greiman

wolfiex, 2021-05-10:
> SdFat/src/SdFatConfig.h

Is there not a way that doesn't require editing the source files? What if you have other projects that do not require this?

greiman, 2021-05-10:
>  I am trying to run the soft serial example script (as provided) on an Arduino Leonardo. 

Do you mean Software SPI?

I used this [Adafruit microSD](https://www.adafruit.com/product/254?gclid=EAIaIQobChMIiKio4OW_8AIVmh-tBh1UzQVkEAQYASABEgLySvD_BwE) breakout. I connected the SD to pins on a Leonardo like this:
```
const uint8_t SD_CS_PIN = 10;
//
// Pin numbers in templates must be constants.
const uint8_t SOFT_MISO_PIN = 12;
const uint8_t SOFT_MOSI_PIN = 11;
const uint8_t SOFT_SCK_PIN  = 13;
```
I edited SdFtConfig.h like this:
`#define SPI_DRIVER_SELECT 2`

The example compiled and ran with this output:
```
Type any character to start
This line was printed using software SPI.
Done.
```