greiman/SdFat #271: Cant get it to work on teensy 4.1

jefjef, 2021-03-03:
Obviously I'm missing something but I don't know what.

This is my code: 

```c++
#include <SdFat.h>

#define SD_CONFIG SdioConfig(FIFO_SDIO)

SdFs sd;

void readFile()
{
    delay(5000);
    Serial.println("TRYING");
    if (!sd.begin(SdioConfig(SD_CONFIG)))
    {
        sd.initErrorHalt(&Serial);
        Serial.println("CANT BEGIN");
        return;
    }
    Serial.println("BEGUN");
}
```

but all I'm getting is

```
TRYING
begin() failed
Do not reformat the SD.
SdError: 0X25,0X1
```

I'm using a 64GB exFat formatted (using SDCarFormatter) SDXC card.

If someone could tell me what I'm doing wrong that would be much appreciated.

stevstrong, 2021-03-03:
Correct would be:
`sd.begin(SD_CONFIG)
`

jefjef, 2021-03-03:
Oh with all the attempts with different configs I probably copy pasted something wrong.
I mate it `sd.begin(SD_CONFIG)` but still get the same error.

greiman, 2021-03-03:
This works with a Teensy 4.1:
```
#include <SdFat.h>

#define SD_CONFIG SdioConfig(FIFO_SDIO)

SdFs sd;

void readFile()
{
    delay(5000);
    Serial.println("TRYING");
    if (!sd.begin(SD_CONFIG))
    {
        sd.initErrorHalt(&Serial);
        Serial.println("CANT BEGIN");
        return;
    }
    Serial.println("BEGUN");
}
void setup() {
  Serial.begin(9600);
  readFile();
}
void loop() {}
```
Output:
```
TRYING
BEGUN
```
Do you have another card to try?

jefjef, 2021-03-03:
Damn I don't, I actually bought this one because an older one I had also wouldn't work.
Thanks though, at least I know now that its my card and not my code.

closed 2021-03-03 by jefjef

jefjef, 2021-03-03:
I have now tried 3 different cards;
-Samsung Evo Plus 6Gb
-Kingston 4Gb SDHC
-SanDisk 1Gb normal MicroSd

And none of them work 😞

jefjef, 2021-03-03:
My complete project is [here](https://github.com/jefjef/00Key/blob/feature/new-menu/firmware/src/Storage.cpp), maybe i configured something wrong in platformio?

greiman, 2021-03-04:
I suspect there may be a problem with the Teensy.  The SD command that failed is the first command that required a return of data using the 4-bit SDIO bus.

The SD bus includes the following signals:
CLK: Host to card clock signal
CMD: Bidirectional Command/Response signal.
DAT0 - DAT3: 4 Bidirectional data signals.

The first few initialization commands use just CLK and CMD.  The command that failed uses DAT0 - DAT3.  If there is a problem with the socket or the PCB for one of the signals, the command will fail due to a CRC error.

You might ask on the [Teensy Forum](https://forum.pjrc.com/forums/3-Technical-Support-amp-Questions) if anyone has seen this problem.

jefjef, 2021-03-04:
Ah, I'll have a closer look at the teensy itself later maybe I damaged something when soldering.
Thanks for the update! 

If I cant get it to work I'll assume its a hardware fault and switch over to storing data in the EEPROM.