greiman/SdFat #272: SD card is not recognized whe using Arduino MKR WiFi 1010

ahmadmadary, 2021-03-12:
Hi

First, i want to thank you for your amazing work with this library. However, I am facing some problems with it.  I am using adafruit microSD together with SdFat library for a project, but i have faced a strange problem.
when I hook up SDcard to Arduino MKR WiFi 1010 and upload "SdInfo" example, i get the following error:

error: cardSize failed
SD errorCode: 0X50,0X20

when i upload the "CardInfo" example from SD library to the same board with the same wiring, it works perfectly. I checked the "SdInfo" with arduino uno, and it works just fine.
so i am sure the problem is not with the wiring or with the SDcard and SD module.

I deeply appreciate it is you can point me to a direction so i can fix this issue.

thank you

greiman, 2021-03-12:
What version of SdFat are you using? The above error code is not in the current version of SdFat.

https://github.com/greiman/SdFat/blob/master/doc/SdErrorCodes.txt

ahmadmadary, 2021-03-12:
that was version 1.1.4 since i originally wrote the code with that version, but when I updated the library to 2.0.5, i got the following error

'mbr_t' was not declared in this scope

I downloaded the Version 2.0.5, and ran the SdInfo, here is the error:

init time: 129 ms
readInfo failed
SD errorCode: SD_CARD_ERROR_READ_TOKEN = 0x18
SD errorData = 0x3a

the same code is working with Arduino Uno.

I ran SdFormatter and got the following error:
error: Get sector count failed.
SD errorCode: SD_CARD_ERROR_READ_TOKEN = 0x18
SD errorData = 0x20

Thanks

greiman, 2021-03-12:
I suspect there is a problem with SPI clock rate.  The old SD.h, which is a wrapper for a 2009 version of SdFat, uses a clock rate of 4 MHz.

Newer versions of SdFat try to use the highest clock rate supported by the board.  The SD is initialized at a max rate of 400 kHz.
This seems to work with your board.  

At the end of init, a higher speed is requested.  The board should use the highest supported rate that is less or equal to the requested rate. This is where your board fails.  I have seen this in some other boards with a SAMD.

Could you try the [QuickStart](https://github.com/greiman/SdFat/blob/master/examples/QuickStart/QuickStart.ino) example? This example uses a max clock rate of 4 MHz.

If QuickStart works try setting a lower rate in SdInfo by editing these lines.
```
#define SD_CONFIG SdSpiConfig(SD_CS_PIN, DEDICATED_SPI, SD_SCK_MHZ(16))
#else  // HAS_SDIO_CLASS
#define SD_CONFIG SdSpiConfig(SD_CS_PIN, SHARED_SPI, SD_SCK_MHZ(16))
```

greiman, 2021-03-12:
Part of the problem may be the adafruit microSD breakout.  This board is great for AVR since it has the required 5V to 3.3V level shifter.  For 3.3V boards like the  Arduino MKR WiFi 1010 there can be problems at higher clock rates.

You could use a simple SD socket with no level shifter and possibly use higher SPI clock rates.

ahmadmadary, 2021-03-13:
Thank you very much. QuickStart worked just fine and I did what you said, lower frequency and now everything works perfectly. thank you very much.
I cannot thank you enough for your quick reply.