greiman/SdFat #273: File open will fail if two or more sequential directories begin with the same substring

AidanHockey5, 2021-03-12:
Quick details:
SD FAT Version is 1.0.3
Fairly old, I know, but I require features in that version that weren't carried over to it's successors. 
MCU: ATSAMD51J19A using an Adafruit Metro M4 bootloader
16GB FAT32-formatted SD card accessed via hardware SPI using SPI_FULL_SPEED

Background: I am developing a music player that drives old video game sound chips. I use SdFat to drive my SD card, and for the most part, it's been working great.

Issue:
If two or more sequential directories begin with the same substring, the SdFat library will actually fail to open files from the latter directories. It's not uncommon to run into a situation like this in my case, as old game soundtracks often have sequels with similar names, for example:

If I have an SD card with the following folders...

```
Shining Force
Shining Force II
```

Notice that both of those folders begin with the same substring, "Shining Force," but the complete directory name strings are different. These are separate directories. Also notice that their location on the SD card is sequential. The original "Shining Force" directory is first, the "Shining Force II" folder is second.

If I was to add files to these directories like so...

```
Shining Force
   |_myFile1.vgm
   |_myFile2.vgm
Shining Force II
   |_myFile3.vgm
   |_myFile4.vgm
```

I'm able to load `myFile1.vgm` and `myFile2.vgm` from the `Shining Force` folder just fine.

However, if I try to open `myFile3.vgm` or `myFile4.vgm` from the `Shining Force II` folder, the library will always fail to open the file. Even if instructed to directly open the file like so...
```
  if(SD.open("/Shining Force II/myFile3.vgm", O_READ))
    Serial.println("File OK");
  else
    Serial.println("FILE OPEN FAILED!");
```

The result will always be "FILE OPEN FAILED!"

Something tells me that SdFat is actually opening the first `Shining Force` folder and looking for myFile3.vgm, even though it doesn't exist in that directory.

If I rename the folders to something like this...

```
1_Shining_Force
2_Shining_Force
```

SdFat will operate normally as the two directories no longer begin with the same substring.

Thank you for your help in advance! Feel free to let me know if you need any more details!

greiman, 2021-03-13:
This problem still is in V 2.  Fortunately I am currently rewriting this part of SdFat since it has many problems so I can give you a patch that should work. I have only tested on an earlier version of SdFat.

At line 444 of SdFat-1.0.3/src/FatLib/FatFileLFN.cpp
replace this line:
```
        lfnOrd = ord = ldir->ord & 0X1F;
```
with: 
```
        ord = ldir->ord & 0X1F;
        if (ord != (freeNeed - 1)) {
          continue;
        }
        lfnOrd = ord;
```
Let me know if it works.

Also what features do you need that prevents upgrade to V2?

AidanHockey5, 2021-03-13:
Yes! Fantastic, thank you so much @greiman, you rule.
For my particular version of SdFat, the problematic line was actually replaced on L344, but the end result works great!

> Also what features do you need that prevents upgrade to V2?

I think it mostly came down to "not broken, don't fix." Lots of the SdFat examples are based around the older versions which have around a decade of support built around it. I think the version I'm on now is a hold-over from previous projects where the library "just worked" without any changes to my codebase. I think my main issue is that support for the `vwd()` function was dropped, but my code leveraged it so much that it would have been a real bear to upgrade. I think a few file operation functions were also deprecated or renamed, but it's been a while since I've last checked so I can't entirely remember what prevented me from upgrading. It very likely a case of "monkey see red squiggles, no upgrade" haha 

Since the library already worked so well, I decided it wasn't worth the effort at the time. Something tells me you have pretty robust alternatives for the items that were deprecated, but truthfully, I haven't put in the time to retool my end of the software to discover them. 
For future projects that don't require any legacy code from my old projects, I'll absolutely be leveraging V2 because exFAT support sounds awesome

Up until now, I've actually had no issues with SdFat and this is the first bug I've ever personally encountered, which is incredible given the complexity of this library. Well done and thanks so much again!

closed 2021-03-13 by AidanHockey5