greiman/SdFat #274: Problem with Mega + W5500 but work OK with W5100

soyferny, 2021-03-14:
Hi

I've a Arduino that download a XML file from an IP address and store on SD Card
When I was using sdFat V1.1.4 there was no problems. After update to V 2.0.5 Arduino start to download the file but suddenly crash and only the fisrt 20 lines are downloaded and stored on SD. From this point Etheret conection doesn't work anymore. With the same Arduino if i replace the Original Ethernet Shield v2 (W5500) for a clone Ethernet shield W5100 all works fine.
With sdFat 2.0.5 I've tested SPI_HALF_SPEED, 4MHz, 2MHz, 1Mhz... but always the same result, the same lines are downloaded.
Any help??

Sorry for my bad english :(

greiman, 2021-03-14:
I have no idea what is happening.  The fact that your program works with the W5100 shield but not the W5500 means the problem is not easy to diagnose. 

Sorry I can't be of more help.

soyferny, 2021-03-17:
Hi again

I've dissabled all DEBUG_PRINT and now works without problems...
Is sdFat library 2.0.5 using Timer5, i use PWM output PINS 44, 45 and 46???

greiman, 2021-03-17:
SdFat uses no timers or interrupts.  It only uses The SPI pins, SD chip-select, MISO, MOSI, and SCK.

soyferny, 2021-03-22:
I found a rally good solution, initialize Serial at 115200 bps
I don't know why, but there is a problem between Mega 2560, W5500 and Serial at 9600 bps, same code with W5100 works but not with W5500. Now (at 115200 bps) working with both shields.

Hope this solution can help to anyone with same problem

Really thanks for your support and your library