greiman/SdFat #275: Not issue but question and minor error?

mjs513, 2021-03-17:
I've been experiment exFAT formatter and noticed that when I format a 64GB SD Card with ExFormatter.ino it looks like it leaves 16 MB Unallocated before the the actual volume.  Was wondering why? 

Oh think I found a minor error in the library.

In exFatFormatter.cpp you have
```
  // Allocate two reserved clusters, bitmap, upcase, and root clusters.
  secBuf[0] = 0XF8;
  for (size_t i = 0; i < 20; i++) {
    secBuf[i] = 0XFF;
  }
```
Think it should probably be:
```
  // Allocate two reserved clusters, bitmap, upcase, and root clusters.
  secBuf[0] = 0XF8;
  for (size_t i = 1; i < 20; i++) {
    secBuf[i] = 0XFF;
  }
```

greiman, 2021-03-18:
First, the for loop should  start with one, not zero.  Legacy FAT tables have FFFFFFF8 as the first entry.

SD cards internal  flash memory is managed in "Allocation Units".  

> AU (Allocation Unit): is a physical boundary of the card and consists of one or more blocks and its size depends on each card. The maximum AU size is defined for memory capacity. Furthermore AU is the minimal unit in which the card guarantees its performance for devices which complies with Speed Class Specification.

The standard SD format for exFAT specifies a BU, Boundary Unit which is a multiple of an AU.  For a 64GB card the BU is 16 MB.  Here is a quote from the spec.

> exFAT file system for SD Memory Card has the similar recommendation with FAT32 file system described in the previous section. The structure of the file system should be implemented as follows. 1. The number of the sectors before Boot Region is the same as the Boundary Unit size. Boundary Unit is the logical value determined taking the recommended size at physical layer into account. The concrete value is described in Appendix C.5.3. 2. The total number of the sectors included in Boot Region and FAT is a multiple size of the Boundary Unit. Here, this size shall be set to minimum complying with the format parameter computations described in Appendix C.5.4. 3. The first sector of the Master Boot Record, the Main Boot Sector, and the first sector of the Cluster Heap are always placed on the boundary of Boundary Unit.

So there are several reserved regions in the card to align file structures on AUs.

stevstrong, 2021-03-18:
I think @mjs513 wanted to point out that your code needs to be corrected , where the loops starts from 0:
https://github.com/greiman/SdFat/blob/4582dd4d6c0b19da8999d62aa77da33a287f80cf/src/ExFatLib/ExFatFormatter.cpp#L202-L206

greiman, 2021-03-18:
@stevstrong Yes I will update the loop to` for (size_t i = 1 ;  i < 20; i++)`.

mjs513, 2021-03-18:
Thanks for the explanation Bill.  Reading the MS exFAT spec I don't remember seeing that mentioned at all.  Guess I have to go look at the SD memory card spec.

Have to say one thing.  The more I look at what you put together the more I want say what an amazing job you did!  I would have gone crazy trying to put something like that together.

Do have a couple of additional questions met to ask - my ignorance in not knowing.

When you initialize the FAT (exformatter)
`  ns = ((clusterCount + 2)*4 + BYTES_PER_SECTOR - 1)/BYTES_PER_SECTOR;
`
and when you do the bitmap:
`  bitmapSize = (clusterCount + 7)/8;  
`
don't understand the +7)/8 and in ns +2/4?   Just learned 1<<x is the same as 2^x thanks to looking at your code :)

greiman, 2021-03-18:
`ns = ((clusterCount + 2)*4 + BYTES_PER_SECTOR - 1)/BYTES_PER_SECTOR;`

The + 2 is because the first two entries in a FAT do not represent clusters.

> The FatEntry[0] field shall describe the media type in the first byte (the lowest order byte) and shall contain FFh in the remaining three bytes. The media type (the first byte) should be F8h

Media type is generally ignored so my bug of not setting it correctly was not caught earlier.

> The FatEntry[1] field only exists due to historical precedence and does not describe anything of interest.

The )*4 is because entries are four bytes.  The expression rounds up to a whole number of sectors.

`bitmapSize = (clusterCount + 7)/8;`

The 7)/8 converts clusterCount to bytes rounded up to whole bytes.

mjs513, 2021-03-18:
Thanks for the explanation Bill.

Be honest kind of skimmed over the FatEntry explanation in the spec :(  was concentrating on where things were located on the volume and sizes of areas.  Back to reading.

Thanks again.

closed 2021-10-01 by mjs513