greiman/SdFat #276: Incorrect freeClusterCount() result?

kostuch, 2021-03-19:
Hi,
I'm using latest version of you r excellent library.
Try to compute free space on the card.
Total size is ok. I do sd.clusterCount() * sd.sectorsPerCluster() / 2000 and get megabytes.
But when I use freeClusterCount() on almost empty 8GB (exFat) card I got:
printf("ClusterCount: %d FreeClusters: %d SecPerCluster: %d BytesPerCluster: %d", sd.clusterCount(), sd.freeClusterCount(), sd.sectorsPerCluster(), sd.bytesPerCluster());

ClusterCount: 243072 FreeClusters: 2430 SecPerCluster: 64 BytesPerCluster: 368

Almost full 2GB (FAT32) card:
ClusterCount: 495216 FreeClusters: -1 SecPerCluster: 8 BytesPerCluster: 4096

How to interpret numbers returned by freeClusterCount()?
Is it a proper method to know free space on the card?
Regards

greiman, 2021-03-19:
I have no idea why you get those results.  Your cards clearly were not formatted with the standard SD format which is required for max performance.

I used Windows 10 to format two similar cards and got these results from the [SdInfo](https://github.com/greiman/SdFat/blob/master/examples/SdInfo/SdInfo.ino) example. 
 
8GB card exFAT 64 sectors per cluster:
```
cardSize: 7948.21 MB (MB = 1,000,000 bytes)
flashEraseSize: 128 blocks
eraseSingleBlock: true

OCR: 0XC0FF8000

SD Partition Table
part,boot,bgnCHS[3],type,endCHS[3],start,length
1,0X0,0X82,0X3,0X0,0X7,0XFE,0XFF,0XC5,8192,15515648
2,0X0,0X0,0X0,0X0,0X0,0X0,0X0,0X0,0,0
3,0X0,0X0,0X0,0X0,0X0,0X0,0X0,0X0,0,0
4,0X0,0X0,0X0,0X0,0X0,0X0,0X0,0X0,0,0

Scanning FAT, please wait.

Volume is exFAT
sectorsPerCluster: 64
clusterCount:      242368
freeClusterCount:  242362
fatStartSector:    10240
dataStartSector:   12288
```
2GB card FAT32 8 sectors per cluster:

```
cardSize: 2032.66 MB (MB = 1,000,000 bytes)
flashEraseSize: 32 blocks
eraseSingleBlock: true

OCR: 0X80FF8000

SD Partition Table
part,boot,bgnCHS[3],type,endCHS[3],start,length
1,0X0,0X3,0X3D,0X0,0XC,0XFE,0X3F,0XF6,249,3969799
2,0X0,0X0,0X0,0X0,0X0,0X0,0X0,0X0,0,0
3,0X0,0X0,0X0,0X0,0X0,0X0,0X0,0X0,0,0
4,0X0,0X0,0X0,0X0,0X0,0X0,0X0,0X0,0,0

Scanning FAT, please wait.

Volume is FAT32
sectorsPerCluster: 8
clusterCount:      495200
freeClusterCount:  495196
fatStartSector:    703
dataStartSector:   8441
```
Do you have a reason to format cards with a non-standard format?  Cards internal flash structure is designed for the standard format produced by the [SD formatter](https://www.sdcard.org/downloads/formatter/).

kostuch, 2021-03-19:
This is odd.
I just used standard 'disk manager' utility on Win10 to create volume and fast format.
I'll try to use linked SD formatter.

kostuch, 2021-03-20:
I reformatted card using dedicated tool and now have reasonable result from freeClusterCount()
Lesson learned: Use proper tools to format :)