greiman/SdFat #277: SdFat.begin fail with STM32

pat1, 2021-03-19:
I have avr 1284p and sd card.
everything works well with SD standard library and SdFat.

I change MCU with STM32F103 with the same sd card.
With the standard SD library  everything works well.

With SdFat library SD.begin(SDCARD_CHIP_SELECT_PIN) return false.

Debugging I found the line:

https://github.com/greiman/SdFat/blob/4582dd4d6c0b19da8999d62aa77da33a287f80cf/src/FatLib/FatPartition.cpp#L418

cause the fail.

I am going to migrate a project to the STM32 mcu and do not use SdFat libary is not easy.
Thanks in advance for any suggestion

greiman, 2021-03-20:
What board support package are you using for STM32?

Try running the [QuickStart](https://github.com/greiman/SdFat/blob/master/examples/QuickStart/QuickStart.ino) example and post an error code if it fails.

pat1, 2021-03-20:
The  QuickStart example works out of the box !
Looking at difference with my test I found:
* SD.begin(SDCARD_CHIP_SELECT_PIN,SPI) do not works
* SD.begin(SDCARD_CHIP_SELECT_PIN,SPI_SPEED) with #define SPI_SPEED SD_SCK_MHZ(4)  works !

I do not know the max speed supported but now everything works. 

My board is a microduino stm32 clone https://wiki.microduinoinc.com/Microduino-Module_CoreSTM32  self made:
https://github.com/r-map/rmap/blob/master/kicad/stima_devel/microduino_stm32/microduino_stm_front.png

Thaks a lot for your suggestion and your library

Paolo

closed 2021-03-20 by pat1

greiman, 2021-03-20:
I have seen this problem before with STM32F103.  The SPI driver should set the clock rate to the highest rate supported by the CPU that is less than or equal to the requested rate.  I request 50 MHz by default.  With a 72 MHz clock, this STM32F103 [datasheet](https://www.st.com/content/ccc/resource/technical/document/datasheet/33/d4/6f/1d/df/0b/4c/6d/CD00161566.pdf/files/CD00161566.pdf/jcr:content/translations/en.CD00161566.pdf) specifies a max rate of 18 MHz.  Some libraries set the rate to 36 MHz for SPI1 which sometimes works but often fails.