greiman/SdFat #278: ESP32 with non-standard SPI pins

manuelmitasch, 2021-03-19:
Hello Bill,

I'm trying to use SdFat with an ESP32 board that has a SD card soldered to non-standard SPI pins (miso: 19, mosi: 13, sck: 14, cs: 15).
I have one board with a working SD card (confirmed with the ESP32 SD library) and a couple boards with presumably corrupted filesystem, that I would like to format with SdFat. I'm testing on the board with the working SD card.

I have read the documentation, github issues and tried to understand the internals of this library, but can't seem to make more progress. HELP!
What I tried so far is using the Quickstart example as a starting point and adding the following lines:

```
#define ENABLE_DEDICATED_SPI 
#define SDCARD_SS_PIN 15

#define SD_MISO 19
#define SD_MOSI 13
#define SD_SCK 14
#define SD_CS SDCARD_SS_PIN
#define SD_EN 16

SPIClass SD_SPI(HSPI);
#define SPI_CLOCK SD_SCK_MHZ(4)
#define SD_CONFIG SdSpiConfig(SDCARD_SS_PIN, DEDICATED_SPI, SPI_CLOCK, &SD_SPI)

void customSetup() {
  pinMode(SD_EN, OUTPUT);
  digitalWrite(SD_EN, LOW);
}
```

The SD_EN is a pin to enabled/disable power to the SD card. Thus, I call the customSetup method at the end of setup().

Also changed sd.begin to use the defined SD_CONFIG:
`  if (!sd.begin(SD_CONFIG)) { 
`

I then edited SdSpiESP.cpp to begin the SPI port with the correct SPI pins:
`m_spi->begin(SD_SCK, SD_MISO, SD_MOSI, SD_CS); //SCK,MISO,MOSI,ss
`

The relevant output is:
```
SPI pins:
MISO: 19
MOSI: 23
SCK:  18
SS:   5

Enter the chip select pin number: 15

SD initialization failed.

errorCode: 0x17, errorData: 0x1
```

The errorCode seems to be: Activate card initialization
Sometimes I get errorCode 0x12 (Read OCR register) at first run.

I don't know with which filesystem the SD card was formatted by the factory. 
So I tried to use all the SD_FAT_TYPE, but everytime it shows the same result.

What else can I try?
Thank you!

greiman, 2021-03-20:
You will need to have someone with ESP32 experience help you setup non-standard SPI pins.  I don't really use ESP32.

manuelmitasch, 2021-03-20:
I have this board + SD card with these pins working with the SD library. 
As the SPI port initialization is the same in both code examples, I would expect it to work. 
Anyway, thank you for your time.

manuelmitasch, 2021-04-05:
My colleague got a solution to this problem. It works when using SoftSPIDriver:

```
const uint8_t SD_CS_PIN = 15;
const uint8_t SOFT_MISO_PIN = 19;
const uint8_t SOFT_MOSI_PIN = 13;
const uint8_t SOFT_SCK_PIN  = 14;

SoftSpiDriver<SOFT_MISO_PIN, SOFT_MOSI_PIN, SOFT_SCK_PIN> softSpi;
#define SD_CONFIG SdSpiConfig(SD_CS_PIN, DEDICATED_SPI, SD_SCK_MHZ(0), &softSpi)

closed 2021-04-05 by manuelmitasch