greiman/SdFat #280: SparkFun Apollo3 Core compile error

adamgarbo, 2021-03-21:
Hi Bill,

Thanks for all your work on the SdFat library.

I've been helping SparkFun test the new Mbed version of their Apollo3 Core for the Artemis. When I attempt to compile any of the SdFat examples or use the library in my own code, I run into the following error during compilation:

```
/Users/adam/Documents/Arduino/libraries/SdFat/src/SpiDriver/SdSpiArtemis.cpp: In member function 'uint8_t SdSpiArduinoDriver::receive(uint8_t*, size_t)':
/Users/adam/Documents/Arduino/libraries/SdFat/src/SpiDriver/SdSpiArtemis.cpp:50:10: error: 'arduino::SPIClass' {aka 'class arduino::HardwareSPI'} has no member named 'transferIn'; did you mean 'transfer'?
   m_spi->transferIn(buf, count);
          ^~~~~~~~~~
          transfer
```

Is this perhaps an Mbed SPI-related issue?

Corresponding issue on SparkFun GitHub: https://github.com/sparkfun/Arduino_Apollo3/issues/370

Cheers,
Adam

greiman, 2021-03-21:
At some point in Nov 2019 there was a transferIn() function  for Apollo3.   transferIn() received a block of data and sent 0XFF bytes over MOSI.  I don't recall how I got SdSpiArtemis.cpp,. I think Nathan Seidle did the original code and I modified is slightly.

I can't just change transferIn()  to transfer() because SD cards are full duplex and look at the data on MOSI for commands.

I can't find the board Nate sent me in Nov 2019.  

You could try the following at about line 49 of SdFat/src/SpiDriver/ 

```
uint8_t SdSpiArduinoDriver::receive(uint8_t* buf, size_t count) {
  memset(buf, 0XFF, count);    // <--- add this line
  m_spi->transfer(buf, count);   // <--- change transferIn to transfer
  return 0;
}
```
Let me know if it works.

adamgarbo, 2021-03-21:
Hi Bill,

Thanks for the quick reply. It looks like the change opened up a can of worms, but the errors are now related to `'__FlashStringHelper'`.

```
In file included from /Users/adam/Documents/Arduino/libraries/SdFat/src/common/PrintBasic.cpp:25:
/Users/adam/Documents/Arduino/libraries/SdFat/src/common/PrintBasic.h:38:2: warning: #warning F() macro defined for non Arduino System [-Wcpp]
 #warning F() macro defined for non Arduino System
  ^~~~~~~
/Users/adam/Documents/Arduino/libraries/SdFat/src/common/PrintBasic.h:70:22: error: reference to '__FlashStringHelper' is ambiguous
   size_t print(const __FlashStringHelper *str) {
                      ^~~~~~~~~~~~~~~~~~~
In file included from /Users/adam/Library/Arduino15/packages/SparkFun/hardware/apollo3/2.0.4/cores/arduino/mbed-bridge/core-api/api/Print.h:24,
                 from /Users/adam/Library/Arduino15/packages/SparkFun/hardware/apollo3/2.0.4/cores/arduino/mbed-bridge/core-api/api/Stream.h:25,
                 from /Users/adam/Library/Arduino15/packages/SparkFun/hardware/apollo3/2.0.4/cores/arduino/mbed-bridge/core-api/api/Client.h:22,
                 from /Users/adam/Library/Arduino15/packages/SparkFun/hardware/apollo3/2.0.4/cores/arduino/mbed-bridge/core-api/api/ArduinoAPI.h:29,
                 from /Users/adam/Library/Arduino15/packages/SparkFun/hardware/apollo3/2.0.4/cores/arduino/mbed-bridge/Arduino.h:17,
                 from /Users/adam/Library/Arduino15/packages/SparkFun/hardware/apollo3/2.0.4/cores/arduino/sdk/ArduinoSDK.h:9,
                 from <command-line>:
/Users/adam/Library/Arduino15/packages/SparkFun/hardware/apollo3/2.0.4/cores/arduino/mbed-bridge/core-api/api/String.h:44:7: note: candidates are: 'class arduino::__FlashStringHelper'
 class __FlashStringHelper;
       ^~~~~~~~~~~~~~~~~~~
In file included from /Users/adam/Documents/Arduino/libraries/SdFat/src/common/PrintBasic.cpp:25:
/Users/adam/Documents/Arduino/libraries/SdFat/src/common/PrintBasic.h:35:7: note:                 'class __FlashStringHelper'
 class __FlashStringHelper;
       ^~~~~~~~~~~~~~~~~~~
/Users/adam/Documents/Arduino/libraries/SdFat/src/common/PrintBasic.h:80:24: error: reference to '__FlashStringHelper' is ambiguous
   size_t println(const __FlashStringHelper *str) {
                        ^~~~~~~~~~~~~~~~~~~
In file included from /Users/adam/Library/Arduino15/packages/SparkFun/hardware/apollo3/2.0.4/cores/arduino/mbed-bridge/core-api/api/Print.h:24,
                 from /Users/adam/Library/Arduino15/packages/SparkFun/hardware/apollo3/2.0.4/cores/arduino/mbed-bridge/core-api/api/Stream.h:25,
                 from /Users/adam/Library/Arduino15/packages/SparkFun/hardware/apollo3/2.0.4/cores/arduino/mbed-bridge/core-api/api/Client.h:22,
                 from /Users/adam/Library/Arduino15/packages/SparkFun/hardware/apollo3/2.0.4/cores/arduino/mbed-bridge/core-api/api/ArduinoAPI.h:29,
                 from /Users/adam/Library/Arduino15/packages/SparkFun/hardware/apollo3/2.0.4/cores/arduino/mbed-bridge/Arduino.h:17,
                 from /Users/adam/Library/Arduino15/packages/SparkFun/hardware/apollo3/2.0.4/cores/arduino/sdk/ArduinoSDK.h:9,
                 from <command-line>:
/Users/adam/Library/Arduino15/packages/SparkFun/hardware/apollo3/2.0.4/cores/arduino/mbed-bridge/core-api/api/String.h:44:7: note: candidates are: 'class arduino::__FlashStringHelper'
 class __FlashStringHelper;
       ^~~~~~~~~~~~~~~~~~~
In file included from /Users/adam/Documents/Arduino/libraries/SdFat/src/common/PrintBasic.cpp:25:
/Users/adam/Documents/Arduino/libraries/SdFat/src/common/PrintBasic.h:35:7: note:                 'class __FlashStringHelper'
 class __FlashStringHelper;
       ^~~~~~~~~~~~~~~~~~~
```

Cheers,
Adam

greiman, 2021-03-21:
Try removing the files SdFat/src/common/PrintBasic.h and SdFat/src/common/PrintBasic.cpp.

They should not be needed for an Arduino compatible system.

adamgarbo, 2021-03-21:
Hi Bill,

That did the trick, thanks.

Will it be possible to make v2.x of the Apollo3 Core compatible with SdFat without needing to manually delete the PrintBasic files? For example, ignore them if `defined(ARDUINO_ARCH_APOLLO3)`? Or is this something the folks at SparkFun can address?

Cheers,
Adam

greiman, 2021-03-22:
I will just delete these files for SdFat on github.  They were for my personal use on RTOS systems that don't define Print.

I will fix SdSpiArtemis.cpp, delete the PrintBasic files, and post a new version of SdFat soon.

I have a more general worry.  I don't understand how the F() macro and  class  __FlashStringHelper were defined in PrintBasic.h.

The  only include files in PrintBasic.h are:
```
#include <stdint.h>
#include <stddef.h>
#include <string.h>
```

I worry about a board package where classes and macros are defined without include files.  It seems like other libraries and applications could have problems.

adamgarbo, 2021-03-22:
Thanks Bill,

A quick update that while the code compiles successfully, it appears that there are some Mbed OS issues that manifest once the code is uploaded.

```
#include <SdFat.h>
#include <SPI.h>

SdFs sd;

void setup() {
  SPI.begin();
  Serial.begin(115200); 
  while (!Serial);
  
  if (!sd.begin(SdSpiConfig(41, DEDICATED_SPI)))
    Serial.println("Warning: microSD initialization failed.");
}

void loop() {}
```

**Output:**
```
11:11:29.353 -> ++ MbedOS Error Info ++
11:11:29.353 -> Error Status: 0x80FF0144 Code: 324 Module: 255
11:11:29.353 -> Error Message: Assertion failed: AM_HAL_STATUS_SUCCESS == am_hal_iom_configure(obj->iom.handle, &obj->iom.cfg)
11:11:29.353 -> Location: 0x23851
11:11:29.353 -> File: mbed-os/targets/TARGET_Ambiq_Micro/TARGET_Apollo3/device/iom_api.c+35
11:11:29.353 -> Error Value: 0x0
11:11:29.353 -> Current Thread: main Id: 0x100046A0 Entry: 0x247F1 StackSize: 0x1000 StackMem: 0x100062E8 SP: 0x100071A4 
11:11:29.353 -> For more info, visit: https://mbed.com/s/error?error=0x80FF0144&tgt=SFE_ARTEMIS_MM_PB
11:11:29.391 -> -- MbedOS Error Info --
```

I suspect this will need to be something SparkFun will have to investigate.

Cheers,
Adam

greiman, 2021-03-22:
The above mods are now in SdFat V 2.0.6.

adamgarbo, 2021-03-22:
Thanks, Bill!

closed 2021-03-22 by adamgarbo