greiman/SdFat #282: Default speed of DEDICATED_SPI

adamgarbo, 2021-03-26:
Hi Bill,

Continuing with the testing of the new Apollo3 Core with SdFat, I've noticed that it doesn't seem to like to initialize using the following code:

```
// Initialize microSD
if (!sd.begin(SdSpiConfig(SD_CS_PIN, DEDICATED_SPI)))
{
  Serial.println("Warning: microSD initialization failed.");
  while (1); 
}
```

The Mbed OS error that's thrown appears to be due to trying to set the speed to above 48 MHz. If `SD_SCK_MHZ()` is not explicitly included, does the library default to 50 MHz?

Cheers,
Adam

greiman, 2021-03-26:
The SdFat defaults to 50 MHz.  SPI libraries should set the rate to the highest rate supported by the board/CPU that is less than or equal to the requested rate.  All Arduino libraries do this.

It's important that a SPI library does this since there is no easy way to know the max possible rate.

Here is the key areas from the official Arduino SAMD SPI library:
```
  SPISettings(uint32_t clock, BitOrder bitOrder, uint8_t dataMode) {
...
#if defined(ARDUINO_ARCH_SAMD)
  // The datasheet specifies a typical SPI SCK period (tSCK) of 42 ns,
  // see "Table 36-48. SPI Timing Characteristics and Requirements",
  // which translates into a maximum SPI clock of 23.8 MHz.
  // Conservatively, the divider is set for a 12 MHz maximum SPI clock.
  #define SPI_MIN_CLOCK_DIVIDER (uint8_t)(1 + ((F_CPU - 1) / 12000000))
#endif
...
  this->clockFreq = (clock >= (F_CPU / SPI_MIN_CLOCK_DIVIDER) ? F_CPU / SPI_MIN_CLOCK_DIVIDER : clock);
```

adamgarbo, 2021-03-26:
Hi Bill,

Thanks for the clarification. I'll close this issue and forward it to @Wenn0101 at SparkFun.

Cheers,
Adam

closed 2021-03-26 by adamgarbo