greiman/SdFat #283: readCsvFile.ino doesn't work

polymer-coder, 2021-03-28:
Below is the output of `QuickStart.ino`. I removed the part where it prompted to enter chipSelect and just hardcoded it as pin 15 (ESP3266) instead. 

```
SPI pins:
MISO: 12 
MOSI: 13
SCK:  14
SS:   15

Be sure to edit DISABLE_CHIP_SELECT if you have
a second SPI device.  For example, with the Ethernet
shield, DISABLE_CHIP_SELECT should be set to 10
to disable the Ethernet controller.

SD chip select is the key hardware option.
Common values are:
Arduino Ethernet shield, pin 4
Sparkfun SD shield, pin 8
Adafruit SD shields and modules, pin 10

Assuming the SD is the only SPI device.
Edit DISABLE_CHIP_SELECT to disable another device.

Card successfully initialized.

Card size: 63865 MB (MB = 1,000,000 bytes)

Volume is FAT64, Cluster size (bytes): 131072

Files found (date time size name):

Success!  Type any character to restart.
```

Below is the output of ReadCsvFile.ino

```
Type any character to start
begin() failed
Do not reformat the SD.
SdError: 0X18,0X0
```

And `SdInfo.ino` gives: 

```
SdFat version: 2.0.4

Assuming the SD is the only SPI device.
Edit DISABLE_CS_PIN to disable an SPI device.

Assuming the SD chip select pin is: 15
Edit SD_CS_PIN to change the SD chip select pin.

type any character to start
init time: 26 ms

Card type: SDXC

Manufacturer ID: 0X3
OEM ID: SD
Product: SC64G
Version: 8.0
Serial number: 0X933706F8
Manufacturing date: 9/2014

cardSize: 63864.57 MB (MB = 1,000,000 bytes)
flashEraseSize: 128 blocks
eraseSingleBlock: true

OCR: 0XC0FF8000

SD Partition Table
part,boot,bgnCHS[3],type,endCHS[3],start,length
1,0X0,0XA,0X9,0X2,0X7,0XFE,0XFF,0XFF,32768,124702720
2,0X0,0X0,0X0,0X0,0X0,0X0,0X0,0X0,0,0
3,0X0,0X0,0X0,0X0,0X0,0X0,0X0,0X0,0,0
4,0X0,0X0,0X0,0X0,0X0,0X0,0X0,0X0,0,0

Scanning FAT, please wait.

Volume is exFAT
sectorsPerCluster: 256
clusterCount:      486992
freeClusterCount:  486986
fatStartSector:    49152
dataStartSector:   65536
```

I changed `SD_FAT_TYPE` to 3 for `ReadCsvFile.ino` and `QuickStart,ino`. Any idea on how to fix this?

greiman, 2021-03-28:
You didn't give me details about your hardware and this is a hardware/SPI problem.  I suspect an EESP8266 from the SPI pin-out.

Try reducing the SPI rate.  ESP SPI libraries don't conform to the Arduino standard:

> The SdFat defaults to 50 MHz. SPI libraries should set the rate to the highest rate supported by the board/CPU that is less than or equal to the requested rate. All Arduino libraries do this.
> 
> It's important that a SPI library does this since there is no easy way to know the max possible rate.

Try using 16 MHz like SdInfo.  Then try higher rates until it fails unless you don't care about performance.

From SdInfo:
```
#if HAS_SDIO_CLASS
#define SD_CONFIG SdioConfig(FIFO_SDIO)
#elif ENABLE_DEDICATED_SPI
#define SD_CONFIG SdSpiConfig(SD_CS_PIN, DEDICATED_SPI, SD_SCK_MHZ(16))
#else  // HAS_SDIO_CLASS
#define SD_CONFIG SdSpiConfig(SD_CS_PIN, SHARED_SPI, SD_SCK_MHZ(16))
#endif  // HAS_SDIO_CLASS
```

polymer-coder, 2021-03-28:
Apologies I'm on the ESP8266 Your suggestion worked! I was able to carry `SD_SCK_MHZ `as high as 39

```
Type any character to start

abc
123
456
7.89

def
-321
654
-9.87

ghi
333
255
5.55

Done
```

> This is a hardware/SPI problem

You figured that out via the error codes or just intuition?

greiman, 2021-03-28:
The error code told me it was a hardware/SPI problem.  QuickStart and SdInfo worked and have lower SPI rates.  In this case it's almost always a problem where the SPI library does not limit the rate to the max the CPU can support.

closed 2021-03-28 by polymer-coder