greiman/SdFat #284: Board/Microcontroller Compatibility

maxbla, 2021-03-29:
Is there a list of boards/microcontrollers officially supported by SdFat? I recently discovered that the ATmega4808/4809 (Arduino Nano Every) deals with SPI registers differently from the ATmega328, which makes this library (or at least the SdInfo example) incompatible. This is not a feature request for any particular device, rather a request for information on device compatibility in general.

greiman, 2021-03-30:
Try using the standard SPI library by editing Arduino/libraries/SdFat/src/SfFatConfig.h.

At about line 115 set SPI_DRIVER_SELECT to one like this.

```
/**
 * If the symbol SPI_DRIVER_SELECT is:
 *
 * 0 - An optimized custom SPI driver is used if it exists
 *     else the standard library driver is used.
 *
 * 1 - The standard library driver is always used.
 *
 * 2 - An external SPI driver of SoftSpiDriver template class is always used.
 *
 * 3 - An external SPI driver derived from SdSpiBaseClass is always used.
 */
#define SPI_DRIVER_SELECT 1  // <<-------- Change from zero to one.
```
I will update SdFat to select the standard SPI library with the  new AVR series.  The custom AVR library dates back to the first version of SdFat in 2008 when the standard library was big and slow.  I may only use the standard SPI library for AVR.

maxbla, 2021-03-30:
Changing `SPI_DRIVER_SELECT` to 1 does indeed allow the SdInfo example to run, working as expected, on my Arduino Nano Every. Thanks. Regarding the SPI default driver, that sounds good to me. Regarding hardware support, would it be safe to say that this library aims to support all [Arduino.cc boards](https://store.arduino.cc/usa/arduino/boards-modules)? At the very least, all [Arduino.cc "core" boards](https://store.arduino.cc/usa/new-home/arduino-core-boards).

greiman, 2021-03-30:
SdFat should support all Arduino boards.  Most boards just use the standard SPI library, AVR and Arsuino Due are the main exceptions.  SdFat has many Teensy users.  The Arduino SD.h library is just a wrapper for a 2009 version  of SdFat.

I

greiman, 2021-04-02:
I bought a Nano Every and ran the [bench](https://github.com/greiman/SdFat/blob/master/examples/bench/bench.ino) example with the standard SPI library.  It is slow.  Here are the results:
```
write speed and latency
speed,max,min,avg
KB/Sec,usec,usec,usec
257.94,2652,1964,1975
257.94,2696,1964,1975

read speed and latency
speed,max,min,avg
KB/Sec,usec,usec,usec
259.08,1984,1956,1965
259.13,1984,1956,1965
```
I fixed the custom AVR SPI driver to support the new series of AVR CPUs with the ATmega4808/4809 type SPI controller.

The custom driver is more than twice as fast:
```
write speed and latency
speed,max,min,avg
KB/Sec,usec,usec,usec
625.78,1492,800,808
625.78,1532,800,808

read speed and latency
speed,max,min,avg
KB/Sec,usec,usec,usec
563.57,908,888,898
563.70,908,888,897
```
The new driver will be in SdFat-beta soon.  The next version of SdFat-beta has major new features so it will be a while before the release version has this driver.