greiman/SdFat #285: v2 cannot use with Particle Boron SPI1

zabaat, 2021-03-30:
Hello, we were able to get this working with version 1 of your library, but with v2 we are unable to init. It looks like the ability to call sd(SPI1) is gone now. How to make this work in V2 with particle? Both code are below.

```
#define IMPLEMENT_SPI_PORT_SELECTION 1
// Quick hardware test for SPI card access.
//
// clang-format off
#include <SPI.h>

#include "Particle.h"
#include "SdFat.h"
#include "sdios.h"

// clang-format on
//
// Set DISABLE_CHIP_SELECT to disable a second SPI device.
// For example, with the Ethernet shield, set DISABLE_CHIP_SELECT
// to 10 to disable the Ethernet controller.
SYSTEM_MODE(MANUAL);
SYSTEM_THREAD(ENABLED);
const int8_t DISABLE_CHIP_SELECT = -1;
//
// Test with reduced SPI speed for breadboards.  SD_SCK_MHZ(4) will select
// the highest speed supported by the board that is not over 4 MHz.
// Change SPI_SPEED to SD_SCK_MHZ(50) for best performance.
#define SPI_SPEED SD_SCK_MHZ(4)
//------------------------------------------------------------------------------
// File system object.
SdFat sd((SPIClass*)&SPI1);

// Serial streams
ArduinoOutStream cout(Serial);

// input buffer for line
char cinBuf[40];
ArduinoInStream cin(Serial, cinBuf, sizeof(cinBuf));

// SD card chip select
int chipSelect;

void cardOrSpeed() {
    cout << F("Try another SD card or reduce the SPI bus speed.\n");
    cout << F("Edit SPI_SPEED in this program to change it.\n");
}

void reformatMsg() {
    cout << F("Try reformatting the card.  For best results use\n");
    cout << F("the SdFormatter program in SdFat/examples or download\n");
    cout << F("and use SDFormatter from www.sdcard.org/downloads.\n");
}

void setup() {
    Cellular.off();
    Serial.begin(9600);

    // Wait for USB Serial
    waitFor(Serial.isConnected, 15000);
    delay(1000);
    cout << F("\nSPI pins:\n");
    cout << F("MISO: ") << int(MISO) << endl;
    cout << F("MOSI: ") << int(MOSI) << endl;
    cout << F("SCK:  ") << int(SCK) << endl;
    cout << F("SS:   ") << int(SS) << endl;
    delay(1000);
    if (DISABLE_CHIP_SELECT < 0) {
        cout << F(
            "\nBe sure to edit DISABLE_CHIP_SELECT if you have\n"
            "a second SPI device.  For example, with the Ethernet\n"
            "shield, DISABLE_CHIP_SELECT should be set to 10\n"
            "to disable the Ethernet controller.\n");
[... 328 lines cut ...]
        cout << F("Use a cluster size of 32 KB for cards larger than 1 GB.\n");
        cout << F("Only cards larger than 2 GB should be formatted FAT32.\n");
        reformatMsg();
        return;
    }
    // Read any extra Serial data.
    do {
        delay(10);
    } while (Serial.available() && Serial.read() >= 0);
    cout << F("\nSuccess!  Type any character to restart.\n");
    while (!Serial.available()) {
        SysCall::yield();
    }
}
```

this outputs 
SD initialization failed.
Do not reformat the card!
Is the card correctly inserted?
Is chipSelect set to the correct value?
Does another SPI device need to be disabled?
Is there a wiring/soldering problem?

errorCode: 0x1, errorData: 0xff

zabaat, 2021-03-30:
Nevermind - found the documentation for multiple spi in the examples. thanks for htis lib!

closed 2021-03-30 by zabaat

TrikkStar, 2021-04-20:
@zabaat  Where in the documentation/examples did you find reference for Multiple SPI? I'm in a similar position but not seeing what you found.

zabaat, 2021-04-20:
> @zabaat Where in the documentation/examples did you find reference for Multiple SPI? I'm in a similar position but not seeing what you found.

I think I divined the information from [this](https://github.com/greiman/SdFat/blob/master/examples/STM32Test/STM32Test.ino ) and possibly a fever dream of desperation.

zabaat, 2021-04-20:
@TrikkStar here is the relevant code snippets to init.
```
#define ENABLE_DEDICATED_SPI 1
// clang-format off
#include <SPI.h>
#include "Particle.h"
#include <SdFat.h> // needs init before the publish queue
// #include <SdCardLogHandlerRK.h>
#include "PublishQueueAsyncRK.h"

#define SPI_SPEED SPI_FULL_SPEED  //SD_SCK_MHZ(18)
// Setup SD card writing settings
#define SD2_CONFIG SdSpiConfig(5, DEDICATED_SPI, SPI_FULL_SPEED, &SPI1)

int initSdCard() {
    if (sdCard.begin(SD2_CONFIG)) {  //if (sdCard.begin(SD_CHIP_SELECT, SPI_FULL_SPEED)) 
....
```

greiman, 2021-04-21:
The html documentation in SdFat/doc has the four  sd.begin() calls  and four SdSpiConfig() constructors.  The problem is SdFat now has so many options and classes that it is difficult find what you need in the documentation.  

Here are screen shots of begin and SdFatConfig from SdFat/doc/html: 

[image]

[image]

There are now over a hundred boards that can use SdFat so it is impossible to test or give examples for all of these.