greiman/SdFat #288: did cwd() displear in v2x

rickyelqasem, 2021-04-05:
I needed to update to later version on sdfat to support other library but when I do cwd doesn't work.. and the ide complains its not a member of file.

greiman, 2021-04-05:
There no longer is a single global cwd object since there are now three basic file/volume types.  Each type has a current working volume and each volume has a working directory.

You can do this:

```
  ExFat sd;
  ExFile file;
 
  // Change exFAT current working volume to sd and volume working directory to "Recipes/Pasta".
  sd.chdir("Recipes/Pasta");

  // Open "Recipes/Pasta/Farfalle.txt" on sd for read.
  file.open("Farfalle.txt");
```

How were you using cwd()?

rickyelqasem, 2021-04-05:
I cannot remember now but I think from my code I was going to the top of the CWD so I can start my file:

 sd.chdir(path);
  file.cwd()->rewind();
  
  filecount = 0;
  while (file.openNext(file.cwd(), O_RDONLY)) {

greiman, 2021-04-05:
Just do this:
```
  File dir;
  File file;

  dir.open(path);
  dir.rewind();

  while (file.openNext(&dir, O_RDONLY)) {
```