greiman/SdFat #291: 'File' does not name a type

emp-98, 2021-04-29:
#include <SdFat.h>
SdFat sd;                                       
File file;
char fileName[] = "datalog.csv";     
#define LOGFILE "datalog.csv"
file = sd.open(fileName, FILE_WRITE);

It gives me the following error: 'File' does not name a type. Thanks in advance.

greiman, 2021-04-29:
So what? 

```
int n;
n = strlen("abc");
void setup() {
  // put your setup code here, to run once:
}
void loop() {
  // put your main code here, to run repeatedly:
}
```
when not in a function gives:

```
sketch_apr29a:2:1: error: 'n' does not name a type

 n = strlen("abc");

 ^
```

arnoson, 2021-08-22:
Im using the exact code you posted (This compiles). With a fresh platformio project and a teensy41 and unfortunately it doesn't compile. It used to in another project also using the teensy41. This is the output I get from `platformio run`:
```bash
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/teensy/teensy41.html
PLATFORM: Teensy (4.13.1) > Teensy 4.1
HARDWARE: IMXRT1062 600MHz, 512KB RAM, 7.75MB Flash
DEBUG: Current (jlink) External (jlink)
PACKAGES:
 - framework-arduinoteensy 1.154.0 (1.54)
 - tool-teensy 1.154.210805 (1.54)
 - toolchain-gccarmnoneeabi 1.50401.190816 (5.4.1) 
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 93 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <SdFat> 2.0.7
|   |-- <SPI> 1.0
Building in release mode
Compiling .pio\build\teensy41\src\main.cpp.o
Compiling .pio\build\teensy41\lib797\SPI\SPI.cpp.o
Compiling .pio\build\teensy41\libcc3\SdFat\ExFatLib\ExFatDbg.cpp.o
Compiling .pio\build\teensy41\libcc3\SdFat\ExFatLib\ExFatFile.cpp.o
Compiling .pio\build\teensy41\libcc3\SdFat\ExFatLib\ExFatFilePrint.cpp.o
Compiling .pio\build\teensy41\libcc3\SdFat\ExFatLib\ExFatFileWrite.cpp.o
Compiling .pio\build\teensy41\libcc3\SdFat\ExFatLib\ExFatFormatter.cpp.o
Compiling .pio\build\teensy41\libcc3\SdFat\ExFatLib\ExFatPartition.cpp.o
src\main.cpp:3:1: error: 'File' does not name a type
 File file;
 ^
src\main.cpp: In function 'void setup()':
src\main.cpp:8:3: error: 'file' was not declared in this scope
   file = sd.open(fileName, FILE_WRITE);
   ^
*** [.pio\build\teensy41\src\main.cpp.o] Error 1
```

arnoson, 2021-08-22:
There has been a Teensyduino update recently (https://github.com/PaulStoffregen/cores/releases/tag/1.54) which also mentions `Sdfat` a few times. Seems teensyduino is now using Sdfat beta, but has somehow renamed it to `Sdfat`, so maybe the Sdfat in Teensyduino Core is conflicting with my installed (not beta) version of Sdfat.

greiman, 2021-08-22:
Yes there are conflicts in Teensy.  It's best to use the Teensy wrapper for SdFat.

You could try the github version of SdFat but you must use a more basic type in place of 'File'.  Try replacing 'SdFat' with 'SdFs' and 'File' with 'FsFile'. 

The type 'File' is a typedef but there are conflicts in Teensy so I don't define 'File' in Teensy.

arnoson, 2021-08-22:
Thank you @greiman for the quick help! 
The `SdFat` type was still working, but using `FsFile` did the trick. I'll have a look into the teensy's sd wrapper, although I wan't my code to work with various arduinos so maybe I just create a custom typedef like `SdFile` and set it to `File` by default und to `FsFile` for the teensy.

greiman, 2021-08-22:
It's becoming difficult to write compatible programs that access SD cards on all Arduino like boards.
 
I wrote the first version of SdFat at the end of 2008 for the ATmega328.  Arduino still uses a 2009 version with a few modifications in its SD.h library.  

Teensy has a virtual filesystem that uses SdFat-beta.  ESP boards use a wrapper for FatFS.  The new "Arduino Pro system"  has a new API based mbed SD libraries.

Each RTOS has its own way of handling SD cards and SD drivers.  Too bad Arduino didn't develop a more advanced system and set the standard for a real IoT system a long time ago.

PaulStoffregen, 2021-10-04:
Sorry, it's my fault "File" broke for SdFat-only usage on Teensy.  I want to fix this for Teensyduino 1.56.  Even though we now has a SD library which is merely a thin wrapper for SdFat, where File does work, I also want to find a way to make all the SdFat examples "just work" on Teensy.  So far I do not know any way which doesn't involve adding Teensy-specific code to SdFat... and I'm going on the assumption that won't be so desirable, right?

greiman, 2021-10-04:
At this point I accept that SdFat will have conflicts on many systems.  I will just tell users too bad you can't use SdFat or maybe some of its features.

The File class was introduced in a 2009 wrapper by SparkFun.  I didn't like the idea, The name File is too generic. I had a SdFile class derived from Print and should have just derived SdFile from Stream.  I added the File class to SdFat because so many programs used the Arduino wrapper.

I am now in a major restructuring of the core of SdFat.  I am removing all Arduino dependencies to produce a "library" that is easy to use on any RTOS and the Raspberry Pi Pico system.  I am testing with mbed, ChibiOS and the Pico RP2040 systems.  I am also making it easier to use the core with other block devices like USB drives. 

The Arduino stuff will just be a very simple template class so I suspect something could be done with Teensy to allow File.

PaulStoffregen, 2021-10-04:
I'm excited to hear you're working on a major redesign.  Really curious to hear more.

But I'm also a little dismayed you're testing on other systems but not Teensy, which today may be the only widely used system actually using SdFat as the default library for SD card and other disk access.

PaulStoffregen, 2021-10-06:
As a quick fix, in the copy of SdFat we're shipping to Teensy users, I've edited all the examples to "#define SD_FAT_TYPE 3".  While this doesn't solve the fundamental conflict, hopefully it will help improve user experience.