greiman/SdFat #293: Webbrowser streamFile class SDfile has no member named "size"

R1Snake, 2021-05-05:
Hi,

i want to download via Webbrowser a file from sd card with your library. 

If I call the method:
`web_Server->streamFile(file, "application/octet-stream");`

I got this error:
C:/Users/XXX/.platformio/packages/framework-arduinoespressif32/libraries/WebServer/src/WebServer.h:141:5: error: 'class SdFile' has no member named 'size'
     _streamFileCore(file.size(), file.name(), contentType);

Ok got it. i create a new function in the webbrowserclass with the size as parameter: 
`web_Server->streamFile(file, file.fileSize(), "application/octet-stream");`

and:
```

template<typename T>
  size_t streamFile(T &file, uint32_t filesize, const String& contentType) {
    _streamFileCore(filesize, file.name(), contentType);
    return _currentClient.write(file);
  }
```

Now I got no compilation error but when I call the function got an error on my browser:
error - file incomplete

R1Snake, 2021-05-05:
I tried this code with the same file stored in my spiff and this works fine. 

I checked: 
The size of file -> same
File is opened -> yes

greiman, 2021-05-05:
You should probably use the File class.  File was not a class in the original SdFat.  It was added to the SD.h wrapper for SdFat and just return the underlying fileSize() function.

Unfortunately every fork or work-alike of SdFat seems to need to rename functions so I no longer add aliases.

R1Snake, 2021-05-05:
I changed my code to:
```
SdFat32 sd;
File tmpFile;

....
if (tmpFile = sd.open(filename.c_str(), O_RDONLY)){
....
}
```

and get this error:
no match for 'operator=' (operand types are 'fs::file' and 'File32C/C++')

greiman, 2021-05-05:
Looks like you are using ESP32 or ESP8266.  The class File is define in the system.  You will have various conflicts with SdFat.  I don't define File in ESP systems due to these conflicts.

Try using these type pairs (SdFat32, File32),  (SdExFat, ExFile),  or (SdFs, FsFile) for support of FAT32/FAT16, exFAT, or both FAT/exFAT.

R1Snake, 2021-05-05:
I use SDFat32 and File32 and the code runs.

But I have still a problem if I want to download a file via Webbrowser. See first post. :(

greiman, 2021-05-05:
Can't help with the web server.

ShadowDrakken, 2021-06-10:
from looking at the code for ESP8266WebServer it looks like what might fix this issue is to either replace File32.fileSize() with File32.size() or to simply abstract File32.size() as an alias to File32.fileSize()

ESP8266WebServer's streamFile() method takes T for the file, and then calls file.size(); which I think is where it's failing. I'm pretty new at C/C++ though and am having trouble making and testing this change myself because I can't find where File32.fileSize() is being defined.

ShadowDrakken, 2021-06-11:
@R1Snake bit of a pain and makes for slow page loading, but I found a workaround where I replace streamFile() with a buffered sendContent(). In theory you could increase the buffer size and send the whole file at once, but it's likely you'll run into memory issues if you do. I believe sendContent() only works with ASCII as well, so no UTF or binary files; needs confirmation.

    File32 file = SD.open(path); // Open it
    char buf[255];

    while (file.available()) {
      int n = file.fgets(buf, sizeof(buf));
      server.sendContent(buf);
    }
    //server.streamFile(file, getContentType(path));

    file.close(); // Then close the file again

R1Snake, 2021-06-11:
Here is my code which works without problems. 

File file = sd_Provider->downloadVBOFile(vboFile); //This function returns a File object
  if (file)
  {

    web_Server->sendHeader("Content-Type", "text/text");
    web_Server->sendHeader("Content-Disposition", "attachment; filename=" + vboFile + ".vbo");
    web_Server->sendHeader("Connection", "close");

    web_Server->streamFile(file, "application/octet-stream");
    file.close();
  }
  else
  {
    web_Server->send(404, "text/plain", "FileNotFound");
  }

greiman, 2021-06-11:
> As an aside, I'm a little surprised that SdFat doesn't attempt to be as compatible with SD as possible to make swapping to the better library fairly seamless without breaking other libraries :\

The Arduino SD.h is a wrapper for a very old version of SdFat.  Each fork of SdFat seems to add or subtract something or not update to fix bugs.  Many of these forks and wrappers make it impossible to be compatible. 

File, File32, ExFile, and FsFile all have size() and fileSize().  The following compiles on an Arduino boards with the current version of SdFat.

```
#include "SdFat.h"
File file;
File32 file32;
ExFile exfile;
FsFile fsfile;
void setup() {
  file.size();
  file.fileSize();
  file32.size();
  file32.fileSize();
  exfile.size();
  exfile.fileSize();
  fsfile.size();
  fsfile.fileSize();
}
void loop() {
}
```

> Sketch uses 11412 bytes (35%) of program storage space. Maximum is 32256 bytes.
> Global variables use 274 bytes (13%) of dynamic memory, leaving 1774 bytes for local variables. Maximum is 2048 bytes.

It may not compile on ESP32 depending which libraries are uses since File is defined in the header FS.h in the ESP32 system.  I do a test for using __has_include and don't define types that would conflict.

greiman, 2021-06-11:
Some older programs  may fail if you use a file type that supports exFAT.  file.size() and file.fileSize() will return type uint64_t for exFAT.  Functions like file.seek(pos) require an argument of type uint64_t.  Serial.print(file.fileSize()) will not compile since print does not support uint64_t.

So make sure you only use FAT32 with old programs.  Edit [SdFatConfig.h](https://github.com/greiman/SdFat/blob/master/src/SdFatConfig.h) and force SDFAT_FILE_TYPE to be 1.