greiman/SdFat #294: SdFat + MCP_CAN_lib

JJGR93, 2021-05-06:
Hi, first of all thank you for your work!

I've been researching and trying to make both Sd and MCP2515 work fine.
The issue is that I must call function sd.begin everytime before reading from CAN, or CAN won't work.

Using SHARED_SPI.

Simplified code:
`#include "mcp_can.h"
#include "SdFat.h"

ExFile file;
SdExFat sd;

MCP_CAN CAN(10);

void setup()
{
  Serial.begin(115200);

  initSD();
  initCANBUS();
}

void loop()
{
  readCANBUS();
  saveSD();
}

void initSD()
{
  sd.begin(53, SD_SCK_MHZ(16));
}

void saveSD()
{
  file = sd.open("test", FILE_WRITE);

  if (file)
  {
    file.print("test");
    file.close();
  }
}

void initCANBUS()
{
  while (CAN_OK != CAN.begin(MCP_ANY, CAN_250KBPS, MCP_8MHZ))
  {
    delay(100);
  }

  CAN.setMode(MCP_NORMAL);
}

void readCANBUS()
{
  //...
}`

greiman, 2021-05-06:
I have no idea what the MCP2515 library does or why there is a conflict.  Cant help you.

JJGR93, 2021-05-06:
Ok, thank you very much anyways!