greiman/SdFat #302: SdFat 2-beta to SdFat 2.0 release

RodEnry, 2021-05-18:
Hi, 
I already have an ESP32 running the SdFat 2 beta library with a dedicated SPI (downloaded around March 2020).
I'm working on a standard microSD class 10 16Gb.
Now, I want to update the library and use the latest SdFat 2.

If I replace the old library with the new one, I can see a lot of error on the declaration of the various files.

--------------------------
#include "SdFat.h"
SdFat SD;
File file;                                   <<<< error: 'file' was not declared in this scope
#define SPI_CLOCK SD_SCK_MHZ(25)
#define SD_CONFIG SdSpiConfig(SD_CS_PIN, DEDICATED_SPI, SPI_CLOCK)
--------------------------

If I replace all the "File" with "File32" everything seems to be fine.

- Is it correct this "workaroud"?
- Which is the the main difference between declaring File and File32? (simply explained)

I already search in the documentation but I can't find a clear reply to my doubts.
Many thanks!
Enry

greiman, 2021-05-18:
The problem is File is defined by ESP32 in some cases.

File is just a typedef for File32.

I have a test for when to define File but it could be improved.  The only safe thing to do on ESP32 is to use File32, ExFile or FsFile.

RodEnry, 2021-05-18:
Hi, 
many thanks for the prompt reply!
I'll use File32 then.

For my application is better to use File32, ExFile or FsFile?

Cheers
Enry

greiman, 2021-05-18:
SdFat and File are typedefs so you should use these pairs:

For Fat16/FAT32 only use SdFat32 and File32.
For exFAT only use SdExFat and ExFile.
For FAT32/FAT16/exFAT, use SdFs and FsFile.

On ESP32 I would use SdFs and FsFile.

Here is how SdFat and File are defined.  SDFAT_FILE_TYPE is defined in SdFatConfig.h.
```
#if SDFAT_FILE_TYPE == 1
/** Select type for SdFat. */
typedef SdFat32 SdFat;
/** Select type for File. */
#if !defined(__has_include) || !__has_include(<FS.h>)
typedef File32 File;
#endif
/** Select type for SdBaseFile. */
typedef FatFile SdBaseFile;
#elif SDFAT_FILE_TYPE == 2
typedef SdExFat SdFat;
#if !defined(__has_include) || !__has_include(<FS.h>)
typedef ExFile File;
#endif
typedef ExFatFile SdBaseFile;
#elif SDFAT_FILE_TYPE == 3
typedef SdFs SdFat;
#if !defined(__has_include) || !__has_include(<FS.h>)
typedef FsFile File;
#endif
typedef FsBaseFile SdBaseFile;
#else  // SDFAT_FILE_TYPE
#error Invalid SDFAT_FILE_TYPE
#endif  // SDFAT_FILE_TYPE
```

I don't know how to improve ESP32.  Depending on which libraries you use very strange things happen.  Maybe I should not define Sdfat or File on ESP32.

RodEnry, 2021-05-20:
Many thanks!
Clear as always

closed 2021-05-20 by RodEnry