greiman/SdFat #306: Question about passing SdFat

tickietackie, 2021-05-26:
Hey, this is a question because of too little C++ knowledge. But actually I can't figure it out with Google so I hope somebody can help may here.

I want to have a class where I will do the SdCard handling for a config file and other text files. Maybe this is the first fault because it's a bad architecture, idk.
Nevertheless I wanted to pass the the initialized SD card to the SdCardHandler class in the constructor to call all the functions on it and stick to dependency injection, avoiding a static variable "initialized" in the class. 

The compiler won't let me do this and says: "Function "SdFs::operator=(const SdFs &)" (implicitly declared)" cannot be referenced (is a deleted function)."

Could somebody give me a hint how to solve this?

```
class SdCardHandler {
 private:
  SdFat SD;
  // static initialized varibale
  //static bool initialized;

 public:
  // constructor
  SdCardHandler(SdFat &initializedSdCard) {
    SD = initializedSdCard;  //<-- Compiler comlpains: Function "SdFs::operator=(const SdFs &)" (implicitly declared)" cannot be referenced (is a deleted function)."
  }

...
```

greiman, 2021-05-27:
SdFat is a typedef for the class SdFs.

The class SdFs is derived from a class FsVolume which has no copy constructor or assignment operator.

tickietackie, 2021-05-27:
Ok, so this means there is now way I can have an initialized SD card as an input for a constructor?
How would you go about doing that?

greiman, 2021-05-27:
> How would you go about doing that?

Due to a custom new operator and internal pointers in FsVolume, you can't assign or use a copy/move constructor so I don't know.

TheNitek, 2021-05-30:
I think what you are trying to do is to have constructor like this:
`SdCardHandler(SdFat &initializedSdCard) : SD(initializedSdCard) {}`

I also use `SdFat &SD;` in my classes, but I am not sure if this is necessary/100% correct.

Full example of how I am solving this: https://github.com/TheNitek/RfidShelf/blob/main/RfidShelf/src/ShelfPlayback.h

greiman, 2021-05-30:
Using a reference and initializing it in the constructor like  @TheNitek should be fine.

I saw 
```
 private:
  SdFat SD;
```
and assumed you wanted an instance of SdFat in the class, not a reference.

closed 2021-05-30 by tickietackie