greiman/SdFat #314: int FatFile::read(void*, size_t)' is inaccessible within this context

Twilight-Logic, 2021-07-13:
When I try to use the read() method on an ifstream or fstream object I get this compile error:

'int FatFile::read(void*, size_t)' is inaccessible within this context

Using getline() returns whole lines and works as expected.

Using get() in a loop like so:

  while (sdinout.get(c)) {
    Serial.print(c);
  }

results in most characters being missed and spurious fragmented output being sent to the serial.

Checking the documentation is would appear that read() is listed as a private function which explains the error, but why is it private? I need to read ALL data as binary byte by byte without stopping on and ignoring on bytes that are considered delimiters. The getline() (and get() if it worked) methods do not seem suitable for this task.

I suppose the other option is to use SdFile myfile and file.fgets(). Some sources suggest this is slow in comparison although I haven't done any tests yet.

So, are the stream methods .get() and .read() not meant to be used?

greiman, 2021-07-13:
`istream& get(char& ch)`  always return *this.  `get(char& ch)` reads one character and stores it in ch if available otherwise, leaves ch unmodified and sets failbit.   The [standard ](https://www.cplusplus.com/reference/istream/istream/get/)says `get(char&)` should also set the eofbit.

Looks like read() had a problem and I never fixed it.  Almost no one uses iostreams so you are the first to catch the missing read.

I will put read() and readsome() on the to do list.  May be a long time until I get to it.  My memory is read had a problem since it must return *this and conflicts with use of the underlying read.

The implementation of iostreams is slower than File and I only kept it for some comparability with SdFat V1.

Twilight-Logic, 2021-07-14:
Thanks for having a look and confirming the problem with read() in fstreams.  I used tha approach because a lot of sources and examples seem to use it, although there may be some overlap/confusion with C/C++ which is different from Arduino. I have now converted my code to use the SdFile object and all the methods seem to work fine there. If there is need to use the << or >> operators, then I can always use ArduinoInStream or ArduinoOutStream.

For whatever reason I only get back a block character back when using get().

I realised I didn't give version numbers, but for reference:

Arduino IDE: 1.8.15
SdFat library: 2.0.7
Linux Mint 19.3, 64-bit

Compiled binary running on an ATMega1284P MCU.

greiman, 2021-07-14:
You must use an int type to make get() work. You must test for failbit to make get(c) work.

I need to do a major change to make read work since size_t read(void*, size_t) is in the parent of ifstream and fstream.

read() becomes ambiguous  and fails in may apps that work with a correct iostreams implementation.  I will need to make the the File object be internal to the fstream types.