greiman/SdFat #316: SdFat 2.0.7 - Reports File as directory on open in loop()

fredcwbr, 2021-07-24:
Hello Greiman,

I'm trying to understand whats happening here. 

The configuration is quite simple :
 - Arduino 1.8.13 
 - SW SPI 
 - ArduinoMega 2560 (Arduino brand)
 -  TFT2.4" LCDSHIELD + SD (https://create.arduino.cc/projecthub/javagoza/2-4-tft-lcd-touch-display-road-test-89f51b)
 - HW111 RTC + Eeprom 
 - DHT22 - Thermo/Humidity

My sketch does run, but with a catch, when I use   open( fn , ... ) on loop() it fails. 
Narrowed down to the open call on FatFile.cpp, where isDir() returns 1 , 

strangely, after doing 

bool FatFile::open(FatFile* dirFile, const char* path, oflag_t oflag) {
  FatFile tmpDir;
  fname_t fname;

  Serial.print(F("FatFile.open isOpen "));
  Serial.print(path);
  Serial.print(" :: ");
  Serial.print(isOpen(), HEX);
  Serial.print(" :: isDir  :: ");
  Serial.println(dirFile->isDir(), HEX);
  
I get the isDir() -->   1 on the print, 
 but the 
  if (isOpen() || !dirFile->isDir()) {

does not fail , .. 

and the sketch move on., with isDir() --> false

Attached the sketch (as .txt as .INO is not allowed) ., 

[Touch_shield_newOK.txt](https://github.com/greiman/SdFat/files/6872968/Touch_shield_newOK.txt)

Would like to know where am I going wrong here. Any help is appreciated.,

Thanks in advance, 

Regards from Brasil.
Frederico C Wilhelms

P.S.: Great job. by the way .. Thanks for it.

greiman, 2021-07-25:
I am not sure what is going on in your program.  I ran this test and get the expected result. 

`file.open(&dir, name, opt)` only succeeds if dir is an open directory and file is closed.
Here is the test:
```
#include "SdFat.h"
SdFat sd;
File dir;
File file;

#define error() {Serial.print("Error.");Serial.println(__LINE__);while(true);}

bool testOpen() {
  Serial.print("file.isOpen(): ");
  Serial.println(file.isOpen());
  Serial.print("dir.isOpen(): ");
  Serial.println(dir.isOpen());  
  Serial.print("dir.isDir(): ");
  Serial.println(dir.isDir());
  bool rtn = file.open(&dir, "file.txt", O_RDWR  | O_CREAT | O_APPEND );
  Serial.print("rtn: ");
  Serial.println(rtn);
  return rtn;
}

void setup() {
  Serial.begin(9600);
  // card is on SS so use default begin.  
  if (!sd.begin()) error();
  // check that "dir" and "notdir "exist.
  if (!sd.exists("dir") || !sd.exists("notdir"))error();
  Serial.println("\nCall with dir is regular file");
  if (!dir.open("notdir")) error();
  if (testOpen()) error();
  dir.close();
  Serial.println("\nCall with both closed");
  if (testOpen()) error();
  if (!dir.open("dir"))error();
  Serial.println("\nCall with dir open and file closed");
  if (!testOpen()) error();
  Serial.println("\nCall with both open");
  if (testOpen())error();
  dir.close();
  Serial.println("\nCall with dir closed and file open");
  if (testOpen())error();
}
void loop() {
}
```
Here is the output:

> Call with dir is regular file
> file.isOpen(): 0
> dir.isOpen(): 1
> dir.isDir(): 0
> rtn: 0
> 
> Call with both closed
> file.isOpen(): 0
> dir.isOpen(): 0
> dir.isDir(): 0
> rtn: 0
> 
> Call with dir open and file closed
> file.isOpen(): 0
> dir.isOpen(): 1
> dir.isDir(): 1
> rtn: 1
> 
> Call with both open
> file.isOpen(): 1
> dir.isOpen(): 1
> dir.isDir(): 1
> rtn: 0
> 
> Call with dir closed and file open
> file.isOpen(): 1
> dir.isOpen(): 0
> dir.isDir(): 0
> rtn: 0

greiman, 2021-07-25:
Maybe I don't understand your intent.

(O_RDWR  | O_CREAT | O_APPEND) will open or create a file.
(O_RDWR  | O_APPEND) will only open existing files.
(O_RDWR  | O_CREAT | O_APPEND | O_EXCL) will only create new files.

fredcwbr, 2021-07-25:
Glad to inform that after changing my original call from
 sprintf_P( fnAtual , PSTR("/L%2.2d%2.2d.CSV") , rtc.getHours(), rtc.getMinutes() );
 if( !logFile.open( fnAtual , O_RDWR  | O_CREAT | O_APPEND ) ) {

to 

 if( !logFile.open( &lDir, fnAtual , O_RDWR  | O_CREAT | O_APPEND ) ) {

as you pointed out, its working.

Thanks for the assistance., 

Frederico Wilhelms

closed 2021-07-25 by fredcwbr