greiman/SdFat #317: Get short file name.

GilesBathgate, 2021-07-29:
Hi. I am using exFat but I want to interface with a serial API in the marlin 3d printer firmware that expects DOS8.3 short file names. https://marlinfw.org/docs/gcode/M023.html
To be clear the printer firmware supports long file names, so writing to the SD card with exFAT is fine, its just the API that needs short file names.

So the question is, Is there a way of getting a short filename in exFAT?

```c++
SdFs sd;
sd.begin(SD_CS, SPI_FULL_SPEED);
FsFile file;
file.open(upload.filename.c_str(), O_CREAT | O_WRITE);
char filename[13];
 file.getName(filename, sizeof(filename));
 file.close();
 ```

greiman, 2021-07-30:
The efFAT filesystem does not support short filenames.  There is only a single directory entry for a 255 character UTF name, no short name directory entry like FAT16/FAT32.

You can use 8.3 uppercase names and getName should return what looks like a FAT16/FA32 short name.

GilesBathgate, 2021-07-30:
So it turned out, although the SdFs/fsFile classes write to the sd card, it was actually formatted in FAT32. If I format the card in exFAT Marlin can't recognise it at all.

closed 2021-07-30 by GilesBathgate

GilesBathgate, 2021-07-30:
For anyone else who stumbles across this the working example is:

```c++
SdFat32 sd;
sd.begin(SD_CS, SPI_FULL_SPEED);
File32 file;
file.open(upload.filename.c_str(), O_CREAT | O_WRITE);
file.write(upload.buf, ...);
char filename[13];
file.getSFN(filename);
file.close();
```